SpinalHDL/SpinalHDL #1467: testbench of USB device

Mina-Lrh, 2024-06-25:
How can use Verilog/system-Verilog testbench to verify the function of USB device?

Dolu1990, 2024-06-25:
Hi, 
Do you mean for this one ?
- https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Libraries/Com/usb_device.html

There is no Verilog / System-verilog testbench implemented.
Instead there is a SpinalSim one : 
- https://github.com/SpinalHDL/SpinalHDL/blob/b6f6a35f54c0ed6bdd6fcb4504b880dc2798de14/tester/src/test/scala/spinal/tester/scalatest/UsbDeviceCtrlTester.scala#L50

Mina-Lrh, 2024-06-27:
Yes, thank you very much! I am trying to embedded this IP into SOC. You have mentioned that it can realize Mass storage function. Can I use it as a CVC device through configure USB Device Requests and Descriptors?

Dolu1990, 2024-06-27:
What is a CVC device ?

Mina-Lrh, 2024-06-28:
Sorry, it's CDC (Communication Device Class).

Dolu1990, 2024-06-28:
So, i don't know much about the USB layer above (including CDC) but fondamentaly i implement all the low level primitives, so it should be possible.
I only used this peripheral through a linux driver, so i didn't had to touch the high level stuff.

Mina-Lrh, 2024-07-01:
Thank you very much! I will try it.