SpinalHDL/SpinalHDL #1473: freeze state

ICJJ, 2024-07-08:
I have meet some error.maybe my fault.
but l dont known what happened

issues here:

`
[Error] Simulation failed at time=10
Exception in thread "main" spinal.sim.SimFailure: Simulation ended in a freeze state, there is nothing to squedule which can make time advance.
	at spinal.sim.SimManager.runWhile(SimManager.scala:338)
	at spinal.sim.SimManager.runAll(SimManager.scala:246)
	at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:614)
	at spinal.core.sim.SimCompiled.doSimUntilVoid(SimBootstraps.scala:583)
	at demoUintSint.demoUintSintMain$.delayedEndpoint$demoUintSint$demoUintSintMain$1(demoUintSintMain.scala:13)
	at demoUintSint.demoUintSintMain$delayedInit$body.apply(demoUintSintMain.scala:9)
	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
	at scala.App$class.main(App.scala:76)
	at demoUintSint.demoUintSintMain$.main(demoUintSintMain.scala:9)
	at demoUintSint.demoUintSintMain.main(demoUintSintMain.scala)

Process finished with exit code 1
`

ICJJ, 2024-07-08:
spinalhdl version: 1.10.2a

Dolu1990, 2024-07-08:
it mean the simulation had nothing to do anymore.
What is your testbench ?

ICJJ, 2024-07-08:
testbench like this

object DemoUintSintMain extends App {
  SpinalConfig(targetDirectory = "rtl/DemoUintSint"
  ).generateVerilog(new DemoUintSint)

  SimConfig.withWave.compile(new DemoUintSint).doSimUntilVoid{
    dut =>
      sleep(10)
      println(dut.d5.toInt)
  }
}

Im just print something in the design with simple example.
Why noting todo will report test fail?
Im dont understand exectly！

g0t00, 2024-07-09:
Looking at the simulation this seems expected.
`doSimUntilVoid` (https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Simulation/bootstraps.html):

> Run the simulation until it is ended by calling either simSuccess() or simFailure(). The main stimulus thread can continue or exit early. As long as there are events to process, the simulation will continue. The simulation will report an error if it gets fully stuck.

Thats exactly whats happening you are sleeping for 10 and then doing nothing. Maybe you want to use `doSim` instead...

ICJJ, 2024-07-09:
if use `doSim()`, sim will always pass.
if use `doSimUnitVoid()`, i need use `simSuccess()` or `simFailure()` to end the simulation.Otherwise,sim will always fail.

closed 2024-07-09 by ICJJ

ICJJ, 2024-07-09:
@Dolu1990

Dolu1990, 2024-07-09:
@ICJJ 
I personnaly do not have any access to VCS software, so somebody with it, and which use VCS would need to do a PR.
(in short, it isn't in my hands)
Maybe you can open a PR ?
That could trigger others to review it and merge if all is good :)