SpinalHDL/SpinalHDL #1476: Observations about AFix

thajohns, 2024-07-11:
I'm currently working on a project involving a lot of fixed-point math, and I noticed a few things that annoyed me about the `AFix` API that I wanted to share.

It would be nice if there were more functions that were bit and `exp`-oriented, similar to `.trim()` but with the power to:
- round, saturate, or truncate
- keep the same max or min value or exponent, or specify either
- specify the number of bits in the result, rather than the number being removed
- specify a `HardType[AFix]` to become

Such functions would help with
- creating `HardType[AFix]`es for component port types etc.
- using e.g. DSP blocks, which need a specific number of bits of input

It seems like there are some inconvenient ways in which `AFix`es do not interact nicely with Scala types -- for example, I can't say `a := 1.0 - b`.

Sometimes it is necessary to deal with the bits directly instead of the `AFix` type. Using `.raw` works well enough, but it would be nice to be able to index the bits by their exponent.

Sometimes I want to truncate/saturate/round, but I know that if I saturate away the MSB, I've done something wrong. (Analogously, it could be that rounding the LSB means I've done something wrong.) Functions like `.truncated(...)`, but which only allow removing bits on one side (otherwise producing an error), would help me detect such mistakes. Another `RoundType` that just causes an error if the LSB needs to be removed would allow the existing `.truncate(...)` to do that.

I'd be interested in hearing other peoples' thoughts on the issues presented above.

Dolu1990, 2024-07-12:
> I'm currently working on a project involving a lot of fixed-point math, and I noticed a few things that annoyed me about the AFix API that I wanted to share.

Thanks for the feedback :D

> It would be nice if there were more functions that were bit and exp-oriented, similar to .trim() but with the power to:

API extentions are welcome (PR), as long as things as things are well named there should be no problems. We also have to be carefull to not create duplication (it is easy to forget about one part of the API is already existing)

> HardType[AFix]es for component port types 

Hmm maybe HardType[AFix] should be used with care, because it kinda force you to "craft" the type in order to access to the AFix parameters.
I think we would need a AFixParam class to allow capturing the AFix val maxRaw: BigInt, val minRaw: BigInt, val exp: Int parameters.

> a := 1.0 - b

One issue here is that the resolution of the substraction is kinda unknown.
Something more like : 
a := 1.0.as(b) - b  could be a possible API ? function "as" meaning "The double value 1.0 with the resolution of b"

So, overall, all the points you bring up are very likely valids.

Let's me explain a bit the "story" of AFix.
So, before AFix there was UFix/SFix, which kinda had a lot of limitation / lack of automation.
The wish with AFix is to track the range of value precisely in a automated manner.  
Thing is that for such library to go well, it need a few people to "eat their own dog food" meaning, having to use the thing they develop.  For instance, the issue with me is that i do not work on DSP related stuff in general.
And so, to move things forward with AFix, it just need people like you, using it, bringing stuff on the table, and commiting a pull request :D

So if you feel like this could work, for each group of functionalities you would like to have, you can open one pull request with a squeleton of an API addition (it doesn't even need to be functional), for instance : 

```scala
class AFix ....
  def miaou(arg : Wuff) = ??? // ??? mean not implemented yet in scala
```

And so that way, we avoid having to split things between github issue and PR, and also we have a code squelton to work with.
Avoid having tooooo many thing in a PR, it makes them lonnnnnnng to merge.

Would that work for you ?

thajohns, 2024-07-12:
> We also have to be carefull to not create duplication (it is easy to forget about one part of the API is already existing)

There were quite a few functions that I only found the second time I looked for them, so I see your point. It doesn't help that the documentation seems a bit sparse on `AFix` methods. I'll be sure to look carefully before contributing anything.

> Hmm maybe HardType[AFix] should be used with care, because it kinda force you to "craft" the type in order to access to the AFix parameters.

I'm not quite sure what you mean by "craft." Do you mean that you have to construct an `AFix` before you can have a `HardType[AFix]`? I'm not sure if there's a big problem with that, since presumably `AFix` will have all the nice functions for creating the types one would want.

> I think we would need a AFixParam class to allow capturing the AFix val maxRaw: BigInt, val minRaw: BigInt, val exp: Int parameters.

I guess that would also work. One of the nice things about `HardType`, as mentioned above, is that I can get access to all the type functions I want by making an `AFix`. For example, if I had `typeA` and `typeB`, and I wanted a type representing the product of such types, I could just say `val typeC: HardType[AFix] = typeA() * typeB()`. Replacing that pattern with a new class could involve reimplementing a bunch of functionality.
Maybe there's a better pattern that I don't see right now.

> a := 1.0.as(b) - b could be a possible API ?

I like this. At the same time, I wonder how much can be inferred about the intended type from its usage. For "nice" numbers like 1.0, it almost makes sense to guess the intended resolution from the number of trailing zeros in the floating point's mantissa. For numbers like π, though, it doesn't make sense because `b` have a higher min exp than π. I can't think of a way to guess the intended type in both cases that doesn't have any footguns. There might be a way to do it for integers, though.

> And so, to move things forward with AFix, it just need people like you, using it, bringing stuff on the table, and commiting a pull request :D

I'm actually quite new to the world of hardware design (from a math/software background), so I don't know if I'm using the "right" paradigms. So maybe some of the things I want aren't necessary or headed in the right direction. Then again, it seems like SpinalHDL is itself an attempt to forge a new paradigm, so maybe it isn't a problem.

> Would that work for you ?

Yes. I'll try to put some time into it one of these weekends (or evenings, if I have the time).

Dolu1990, 2024-07-15:
> I'm not quite sure what you mean by "craft." Do you mean that you have to construct an AFix before you can have a HardType[AFix]? 

by "craft" i mean that when you have a HardType instance, if you want to access the parameter of its afix (if you don't have a signal of that HardType, you need to : 
```scala
val xxx = myHardType().maxRaw
```
 myHardType() is equivalent to  myHardType.craft()

> val typeC: HardType[AFix] = typeA() * typeB()

The only downside with this aproache, is that it will create temporary signal in the netlist (which will be removed later on automaticaly). So it has some elaboration time overhead, but that isn't that bad.

> I like this. At the same time, I wonder how much can be inferred about the intended type from its usage.

Ahh i would say, that is to tricky and introduce some kinda of trickyness. Maybe better to always behave the same. Also SpinalHDL can't figure out if you just enter an floating point literal or a floating point value which is the result of some calculation and not precise.
For integer, that would be fine, as they have an total precision.

>  Then again, it seems like SpinalHDL is itself an attempt to forge a new paradigm, so maybe it isn't a problem.

Right, there is a lot of experimentation there. The main thing are : 
- To not break legacy API
- Trying to have explicit API

Do not wait something to be added, just add it through a PR while you dev things ^^
When you feel the need of something is the best time to do it.

> Yes. I'll try to put some time into it one of these weekends (or evenings, if I have the time).

Nice ^^
Note it is summer, i will have some delay some of those weeks (holiday + GF dragon ^^)