SpinalHDL/SpinalHDL #1477: Unnecessary resizing of multiplicands when multiplying `AFix`es

thajohns, 2024-07-11:
A multiplication between two `AFix`es produces Verilog that multiplies two numbers that don't have the original widths of the `AFix`es (or even one more bit for turning unsigned numbers into signed ones). Instead, the multiplicands end up being very wide.
That prevents the FPGA toolchain that I'm working with from inferring DSP blocks for multiplication.

For example, here's some SpinalHDL code:
```scala
case class AFixMult() extends Component {
  val io = new Bundle {
      val a = in(AFix.U(10 bits))
      val b = in(AFix.U(10 bits))
      val c = out(AFix.U(20 bits))
  }
  io.c := io.a * io.b
}
```
and here's the Verilog output:
```verilog
module AFixMult (
  input  wire [9:0]    io_a,
  input  wire [9:0]    io_b,
  output wire [19:0]   io_c
);

  wire       [29:0]   _zz_io_c;
  wire       [29:0]   _zz_io_c_1;
  wire       [19:0]   _zz_io_c_2;
  wire       [9:0]    _zz_io_c_3;

  assign _zz_io_c = _zz_io_c_1;
  assign _zz_io_c_1 = (_zz_io_c_2 * io_b);
  assign _zz_io_c_3 = io_a;
  assign _zz_io_c_2 = {10'd0, _zz_io_c_3};
  assign io_c = _zz_io_c[19 : 0];

endmodule
```
The simplest way to do this is a 10-bit by 10-bit multiplication, but `_zz_io_c_2`, one of the multiplicands, is 20 bits.

Dolu1990, 2024-07-12:
Hi,

So looking at the AFix mul implementation, i can see 

```scala
(_l.asUInt.resize(ret.bitWidth) * _r.asUInt)
```

I'm realy not sure why there is this resize. it seems unnecessary to me aswell.

Dolu1990, 2024-07-12:
Fix inflight https://github.com/SpinalHDL/SpinalHDL/pull/1480

commit 77aa0c96 (2024-07-12): Merge pull request #1480 from SpinalHDL/AFix-dev

Fix #1477 : AFix.* remove unecessary resize
closed 2024-07-12 by Dolu1990