SpinalHDL/SpinalHDL #1478: [feature request] Option to use SystemVerilog unpacked arrays for IO ports

thajohns, 2024-07-11:
Although SpinalHDL `Vec`s work well for interfacing between SpinalHDL components, they produce a bunch of named ports in Verilog modules, which can make it really tedious to interoperate with existing Verilog code. It would be nice if there was a way to produce a Verilog unpacked array as a module port from a `Vec` in a component's `io` bundle.

Dolu1990, 2024-07-15:
Hi,
I think this as been done to some degree.
For blackboxes io definitions, you can use https://github.com/SpinalHDL/SpinalHDL/issues/1258
For the toplevel io definition, i think there is nothing in place. Overall this kind of things can be acheived by post processing things during elaboration time.

Can you give a example of Component and the wished generated io ?

thajohns, 2025-04-01:
Oops, I forgot to reply to this.
I am talking about the toplevel IO ports. The motivation for this question involves using SpinalHDL modules from an existing Verilog codebase.
For example,
```scala
case class TrivialExample() extends Component {
  val io = new Bundle {
      val a = in(Vec.fill(8)(UInt(8 bits)))
      val b = out(Vec.fill(8)(UInt(8 bits)))
  }
  io.b := io.a
}
```
produces a module with the following signature:
```verilog
module TrivialExample (
  input  wire [7:0]    io_a_0,
  input  wire [7:0]    io_a_1,
  input  wire [7:0]    io_a_2,
  input  wire [7:0]    io_a_3,
  input  wire [7:0]    io_a_4,
  input  wire [7:0]    io_a_5,
  input  wire [7:0]    io_a_6,
  input  wire [7:0]    io_a_7,
  output wire [7:0]    io_b_0,
  output wire [7:0]    io_b_1,
  output wire [7:0]    io_b_2,
  output wire [7:0]    io_b_3,
  output wire [7:0]    io_b_4,
  output wire [7:0]    io_b_5,
  output wire [7:0]    io_b_6,
  output wire [7:0]    io_b_7
);
```
... but that's inconvenient to instantiate from a handwritten Verilog module, and does not work well when e.g. the number of ports changes often (which is happening to me at work due to component failure).
So, it would be nice to have something like this instead:
```verilog
module TrivialExample (
  input  wire [7:0]    io_a[8],
  output wire [7:0]    io_b[8]
);
```

PeterWang-dev, 2025-06-21:
Any update on this?

Dolu1990, 2025-06-24:
No update