SpinalHDL/SpinalHDL #1479: how to easily repeat a singal and generates something like {32{awire}}? awire #*32 gives werid verilog

soartang, 2024-07-12:
I have encountered such issues several times. 
I need scala to do something like {32{awire}} & data, but following code gives werid verilog
`
val eInsts     = RegAndFifos.filter(!_.allIsNA).sortBy(_.addr)
reg_rderr := eInsts.map{ x => x.readBits & (x.hitReadAddress #* busif.busDataWidth)}.reduceLeft(_ | _)
`
I'm expecting scala to give me verilog like
`
x0 & {32{hitReadAddress0}} || x1 & {32{hitReadAddress1}} || x2 & {32{hitReadAddress2}}
`
But it gives super super long rtl code. 

So my question is how to easily repeat a signal that produces verilog: {Integer {awire} }?

Dolu1990, 2024-07-12:
Hi,
So far it was not implemented.
But with https://github.com/SpinalHDL/SpinalHDL/commit/9232fba7ec4a7fdf503bcf60024a3300772ecbac it should be good.
I will merge it in dev when the regressions tests are done.

Cheers ^^

Dolu1990, 2024-07-17:
Merged

closed 2024-07-17 by Dolu1990