SpinalHDL/SpinalHDL #1483: Running doSim more than once causes an error and will hang Verilator simulation indefinitely

rpls, 2024-07-16:
I stumbled upon an error with newer SpinalHDL / Verilator versions. Not quite sure yet what change caused this, but this worked perfectly in the past (at least with 1.9 I think).
I usually program my unit tests for components like this:
 
```scala
package example

import spinal.core._
import spinal.core.sim._
import spinal.lib._

import org.scalatest.flatspec.AnyFlatSpec

class TestComp(val width: Int) extends Component {
  val value = out UInt(width bit)
  val cnt = Counter(width bit)
  value := cnt
}

class TestCompTest extends AnyFlatSpec {

  var compiled: SimCompiled[TestComp] = null

  "Component" should "compile" in {
    compiled = SimConfig.compile(new TestComp(4))
  }

  it should "test one thing" in {
    compiled.doSim("firstrun") { dut =>
      dut.clockDomain.forkStimulus(10)
      // do some test...
      dut.clockDomain.waitSampling(10)
    }
  }

  it should "test another thing" in {
    compiled.doSim("secondrun") { dut =>
      dut.clockDomain.forkStimulus(10)
      dut.clockDomain.waitSampling(10)
    }
  }
}
```

I.e., I compile the thing once and do a separate simulation for each behavior I want to test. With Spinal 1.10+, the first run will work fine, but the second run will only print the following and then hang indefinitely:
```
%Error: Adding model when time is non-zero. ... Suggest check time(), or for restarting model use a new VerilatedContext
Aborting...
Warning: Unable to read from client, please check on client for further details of the problem.
```
I think the second line is an output from `sbt`.

This used to work just fine, but causes a hang now. The IVerilog backend is not affected, not sure about the others (but I suspect all other VPI based backends should be fine.

rpls, 2024-07-16:
I did some more testing. This is probably a side-effect of the fix for #1461. I rolled back my `oss-cad-suite` tools to `v5.024-69-g7c9fa8647`, which seems to be a version unaffected by the bug in #1461, and everything is fine from `1.9.4` to `1.10.2a`.

Readon, 2024-07-17:
Personally, I recommend you do not use the latest version.

KireinaHoro, 2024-08-05:
I'm not sure if the behaviour introduced in https://github.com/SpinalHDL/SpinalHDL/commit/b2912435d1c154575915fa722b8061ac2f28d6d6 caused this issue.  The correct fix (IMO) for newer Verilator versions is already merged in https://github.com/SpinalHDL/SpinalHDL/pull/1487, so maybe @rpls you can test again with upstream Verilator 5.026 and `dev` branch of SpinalHDL?

PR #1757 (2025-07-05): fix(sim): Fix EXCEPTION_ACCESS_VIOLATION when calling doSim multiple times with Verilator v5.x+

Closes #1714
Closes #1483 

# Context, Motivation & Description

## Problem
Calling `doSim` multiple times on a single `SimCompiled` instance caused a fatal `EXCEPTION_ACCESS_VIOLATION` with Verilator v5.x+. This regression made it impossible to run multiple simulation sessions on the same compiled design, which is a common pattern in SpinalHDL testing.

## Root Cause
The issue stems from Verilator 5.x's stricter resource management around `VerilatedContext`. In the previous implementation, each `S [... cut]
closed 2025-07-05 by Readon