SpinalHDL/SpinalHDL #1488: Allow bypass of Payload in same stage as insert.

NikLeberg, 2024-07-25:
When I try to bypass a payload in the same stage that it was  inserted into the pipeline, then I get an  `assignment overlap` error.

Altough I know why this happens... does somebody have a working example of how to acheive it nonetheless?
My use case would be to have a payload normally be generated/inserted by one scala module/area, and then under some circumstances overwritten by another value from another moduke/area.

Example:
```scala
import spinal.core._
import spinal.lib.misc.pipeline._

case class Feature() extends Component {
  val A = Payload(Bool())

  val n0, n1 = CtrlLink()
  val l0 = StageLink(n0.down, n1.up)

  val area0 = new n0.Area {
    A := False
  }

  // this bypass gives error
  val area1 = new n0.Area {
    when(True) {
      bypass(A) := True
    }
  }

  // this bypass is fine
  val area2 = new n1.Area {
    when(True) {
      bypass(A) := True
    }
  }

  Builder(n0,  n1, l0)
}

object FeatureTop extends App {
  SpinalConfig().generateVhdl(Feature())
}
```

NikLeberg, 2024-07-27:
Nevermind. I solved it by not using `bypass` at all. If I simply assign to `A` in the other conditional sope, it generates the code I was looking for.

```scala
import spinal.core._
import spinal.lib.misc.pipeline._

case class Feature() extends Component {
  val A = Payload(Bool())

  val n0, n1 = CtrlLink()
  val l0 = StageLink(n0.down, n1.up)

  val area0 = new n0.Area {
    A := False
  }

  val area1 = new n0.Area {
    when(True) {
      A := True
    }
  }

  Builder(n0,  n1, l0)
}

object FeatureTop extends App {
  SpinalConfig().generateVhdl(Feature())
}
```

closed 2024-07-27 by NikLeberg

Dolu1990, 2024-07-27:
here is maybe one solution to the initial proposal : 

```scala
  val area0 = new n0.Area {
    up(A) := False  // instead of A := False
  }
```

(Bypass assume that up(A) exists)