SpinalHDL/SpinalHDL #1496: no waveforms displayed when using xsim

shjdgwj, 2024-07-31:
When simulating using the XSIM as backend, no waveforms displayed after  opening the wdb file
[image]
But it seems like the simulation is done successfully because there debug info can be seen in the console.
[image]
And there isn't any problem when I use Verilator as backend
I don't know how to solve this problem. The simulation code is like:
```
   def spinal = SpinalConfig(
      targetDirectory = "hw/gen",
      defaultConfigForClockDomains = ClockDomainConfig(
        resetActiveLevel = LOW
      ),
      onlyStdLogicVectorAtTopLevelIo = true
    )
    def sim: SpinalSimConfig = SimConfig.withConfig(spinal).withXSim.withXilinxDevice("xczu48dr-fsvg1517-2-e")
...
    dut.doSim("axis_vfifo_sim", 1145141) {
...
}
```

shjdgwj, 2024-07-31:
[axi_vfifo.zip](https://github.com/user-attachments/files/16443769/axi_vfifo.zip)
Here's my wdb file

aureoleday, 2024-08-08:
Same problem. The simulation process seems OK, but the generated wdb file has no waveform data inside

aureoleday, 2024-08-08:
Solved! Need to add .withWave in sim

Dolu1990, 2024-08-08:
Hoooo nice catch XD
Nice ^^

shjdgwj, 2024-08-08:
> Solved! Need to add .withWave in sim

Thank you very much! It works!

closed 2024-08-12 by shjdgwj