SpinalHDL/SpinalHDL #1501: XSIM backend not working on Linux

ramielkhatibPQS, 2024-08-06:
I tried to run the following simple example:

```scala
import spinal.core._

// Identity takes n bits in a and gives them back in z
class Identity(n: Int) extends Component {
  val io = new Bundle {
    val a = in Bits(n bits)
    val z = out Bits(n bits)
  }

  io.z := io.a
}

import spinal.core.sim._

object TestIdentity extends App {
  // Use the component with n = 3 bits as "dut" (device under test)
  SimConfig.withWave.withXSim.compile(new Identity(3)).doSim{ dut =>
    // For each number from 3'b000 to 3'b111 included
    for (a <- 0 to 7) {
      // Apply input
      dut.io.a #= a
      // Wait for a simulation time unit
      sleep(1)
      // Read output
      val z = dut.io.z.toInt
      // Check result
      assert(z == a, s"Got $z, expected $a")
    }
  }
}
```

It is from spinalhdl docs. I only added `withXSim`. I am getting the following error (replaced project path with ${PATH_TO_PROJECT}):
```
source {"${PATH_TO_PROJECT}/./simWorkspace/Identity/xsim/spinal_xsim.tcl"}
couldn't read file ""${PATH_TO_PROJECT}/./simWorkspace/Identity/xsim/spinal_xsim.tcl"": no such file or directory
```

The file does exist, but the problem is that this string is getting double quoted when passed to vivado. I narrowed the problem to #1246 which was fixing XSim for windows. The problem is in [line 243](https://github.com/SpinalHDL/SpinalHDL/pull/1246/files#diff-279ddd318041d78f9dd90c955e9bdc0ca9474da6f2e55d77aa075a22bd962c90R243):

```scala
val command = s""" vivado -mode batch -source "$vivadoScriptPath" """.trim
```

The original code did not have a quote around `$vivadoScriptPath`

Readon, 2024-10-02:
I can confirm this on my side on Linux.

After this have been fixed, there are more problems that have been met on my Linux box. 
With Vivado 2022.1, I have met 
``` bash
setEnvAndRunCmd.sh: 4: exec: compile.sh: not found
```

This can be solved by changing [this line](https://github.com/SpinalHDL/SpinalHDL/blob/0b1d1e9222755cda45793bfa2403b48cbf5c0936/sim/src/main/scala/spinal/sim/XSimBackend.scala#L257) to a full path as below.
``` scala
s"${simulatePath}/${cmd}.sh"
```

The last step is to set the variable by
``` bash
export LD_LIBRARY_PATH=$VIVADO_HOME/lib/lnx64.o
```

Readon, 2024-10-03:
> Thanks for making the pull request! I wish I had time to work on this. I have a suggestion to avoid exporting `LD_LIBRARY_PATH`. You can add it to the cflags here:
> 
> https://github.com/SpinalHDL/SpinalHDL/blob/0b1d1e9222755cda45793bfa2403b48cbf5c0936/sim/src/main/scala/spinal/sim/XSimBackend.scala#L333-L343
> 
> You can add the following:
> 
> ```scala
> s"$vivadoHome/lib/${if(isWindows)"win64.o" else "lnx64.o"}"
> ```

This does not work, as linker report "lib/lnx64.o: file format not recognized".

Do you mean the argument "-L" of GCC to specify the library search directory? I think that is a ldflags part.
That would not affect the runtime searching directory, so does not work too.

ramielkhatibPQS, 2024-10-03:
Sorry, that was a typo (forgot to add -L). I did some testing. It seems that the GCC command is NOT the one that is using the libraries in that path. Instead, the JVM is using it directly in here:

https://github.com/SpinalHDL/SpinalHDL/blob/0b1d1e9222755cda45793bfa2403b48cbf5c0936/sim/src/main/scala/spinal/sim/XSimBackend.scala#L363-L366

The way to get it working is to add the path to `java.library.path` at runtime, but it seems [there is no consistent way to do this across different java versions](https://stackoverflow.com/questions/15409223/adding-new-paths-for-native-libraries-at-runtime-in-java)

Readon, 2024-10-04:
> Sorry, that was a typo (forgot to add -L). I did some testing. It seems that the GCC command is NOT the one that is using the libraries in that path. Instead, the JVM is using it directly in here: [...]

so using environment variable is the best consistent way.

PR #1552 (2024-10-08): fix xsim on Linux.

Closes #1501

Still needs to be tested in different platforms and different Vivado versions.

- [ ] MacOS
- [x] Linux
- [x] Windows
- [x] MSys2

# Context, Motivation & Description

# Impact on code generation

# Checklist

- [ ] Unit tests were added
- [ ] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RTD](https://github.com/SpinalHDL/SpinalDoc-RTD/issues/new?title=Document%20XXX&body=Do%20not%20merge%20until%20SpinalHDL/SpinalHDL%23XXX%20has%20n [... cut]
closed 2024-10-08 by Readon