SpinalHDL/SpinalHDL #1508: LATCH DETECTED from the combinatorial signal

Sanchit-Gupta10, 2024-08-19:
[error] LATCH DETECTED from the combinatorial signal (toplevel/data_ram1/io_rwValid1 : in Bool), defined at
[error]     projectname.Data_RAM$$anon$1.<init>(Data_RAM.scala:9)
[error]     projectname.Data_RAM.<init>(Data_RAM.scala:7)
[error]     projectname.NTT.<init>(NTT.scala:26)
[error]     projectname.MyTopLevelVerilog$.$anonfun$new$73(NTT.scala:743)
[error] ********************************************************************************
[error] ********************************************************************************
[error] LATCH DETECTED from the combinatorial signal (toplevel/data_ram1/io_rwAddress1 : in UInt[7 bits]), defined at
[error]     projectname.Data_RAM$$anon$1.<init>(Data_RAM.scala:10)
[error]     projectname.Data_RAM.<init>(Data_RAM.scala:7)
[error]     projectname.NTT.<init>(NTT.scala:26)
[error]     projectname.MyTopLevelVerilog$.$anonfun$new$73(NTT.scala:743)
[error] ********************************************************************************
[error] ********************************************************************************
[error] LATCH DETECTED from the combinatorial signal (toplevel/data_ram1/io_rwData1 : in UInt[16 bits]), defined at
[error]     projectname.Data_RAM$$anon$1.<init>(Data_RAM.scala:11)
[error]     projectname.Data_RAM.<init>(Data_RAM.scala:7)
[error]     projectname.NTT.<init>(NTT.scala:26)
[error]     projectname.MyTopLevelVerilog$.$anonfun$new$73(NTT.scala:743)
[error] ********************************************************************************
[error] ********************************************************************************
[error] LATCH DETECTED from the combinatorial signal (toplevel/data_ram1/io_rwValid2 : in Bool), defined at
[error]     projectname.Data_RAM$$anon$1.<init>(Data_RAM.scala:15)
[error]     projectname.Data_RAM.<init>(Data_RAM.scala:7)
[error]     projectname.NTT.<init>(NTT.scala:26)
[error]     projectname.MyTopLevelVerilog$.$anonfun$new$73(NTT.scala:743)
[error] ********************************************************************************
[error] ********************************************************************************
[error] LATCH DETECTED from the combinatorial signal (toplevel/data_ram1/io_rwAddress2 : in UInt[7 bits]), defined at
[error]     projectname.Data_RAM$$anon$1.<init>(Data_RAM.scala:16)
[error]     projectname.Data_RAM.<init>(Data_RAM.scala:7)
[error]     projectname.NTT.<init>(NTT.scala:26)
[error]     projectname.MyTopLevelVerilog$.$anonfun$new$73(NTT.scala:743)
[error] ********************************************************************************
[error] ********************************************************************************
[error] LATCH DETECTED from the combinatorial signal (toplevel/data_ram1/io_rwData2 : in UInt[16 bits]), defined at
[error]     projectname.Data_RAM$$anon$1.<init>(Data_RAM.scala:17)
[error]     projectname.Data_RAM.<init>(Data_RAM.scala:7)
[error]     projectname.NTT.<init>(NTT.scala:26)
[error]     projectname.MyTopLevelVerilog$.$anonfun$new$73(NTT.scala:743)
[error] ********************************************************************************

Sanchit-Gupta10, 2024-08-19:
the code for data_ram is 
package projectname

import spinal.core._
import spinal.lib._

case class Data_RAM() extends Component {
  val io = new Bundle {
    // Port 1
    val rwValid1     = in  Bool()
    val rwAddress1   = in  (UInt(7 bits))
    val rwData1      = in  (UInt(16 bits))
    val rwDataOut1   = out (UInt(16 bits))

    // Port 2
    val rwValid2     = in  Bool()
    val rwAddress2   = in  (UInt(7 bits))
    val rwData2      = in  (UInt(16 bits))
    val rwDataOut2   = out (UInt(16 bits))
  }

  // Define the memory with 128 words of 16 bits each
  val mem = Mem(UInt(16 bits), wordCount = 128)

  // Default values to prevent latches
  io.rwDataOut1 := U(0, 16 bits)
  io.rwDataOut2 := U(0, 16 bits)

  // Port 1 Logic
  when(io.rwValid1) {
    // Write operation
    mem.write(
      enable  = True,
      address = io.rwAddress1,
      data    = io.rwData1
    )
  } .otherwise {
    // Read operation
    io.rwDataOut1 := mem.readSync(
      enable  = True,
      address = io.rwAddress1
    )
  }

  // Port 2 Logic
  when(io.rwValid2) {
    // Write operation
    mem.write(
      enable  = True,
      address = io.rwAddress2,
      data    = io.rwData2
    )
  } .otherwise {
    // Read operation
    io.rwDataOut2 := mem.readSync(
      enable  = True,
      address = io.rwAddress2
    )
  }
}

Dolu1990, 2024-08-23:
Hi,

The Mem api to create ports on the memory work in 2 mods.

Either you specify the enable, but you shouldn't use them in a conditional context (when statement)
either you don't specify the enable, and you can use them in a conditional context.

Sanchit-Gupta10, 2024-08-24:
Thanks.
I have modified it and now it is working.

closed 2024-08-24 by Sanchit-Gupta10