SpinalHDL/SpinalHDL #1509: XSim in spinalhdl fail to simulate IPs like FFT or complex multiplier, but works fine in Vivado

aureoleday, 2024-08-20:
I've tested xilinx IPs like multiplier, complex multiplier, divider and FFT in spinal-sim using XSim, most of them work just fine. 
But complex multiplier and FFT seems not working properly. I change the Multiplier Construction method form Use Mults to Use LUTs, then it works OK, but FFT no change however I switch between the construction methods.
Any suggestions on problem?
BTW, those IPs works OK in Vivado Enviroment.

Dolu1990, 2024-08-23:
No idea, i'm only using verilator sim backend

aureoleday, 2024-08-27:
> No idea, i'm only using verilator sim backend

Is verilator able to simulate Xilinx IPs？

Dolu1990, 2024-08-27:
very likely no. won't work if they have any non synthetisable statement

Readon, 2024-10-15:
do you have minimum example?