SpinalHDL/SpinalHDL #1510: Assign bits (BigInt) back to Bundle/Union/Enum in SpinalSim

KireinaHoro, 2024-08-21:
In SpinalSim we now have the capability to interact with buses like AXI to read bits from the design.  My design currently exposes some SpinalHDL `Bundle`, `Union` and `SpinalEnum` over AXI and the simulation testbench needs to parse the fields.  However, creating a signal and using `assignFromBits` wouldn't work since these are elaboration-time APIs.  Can we support this somehow?

Dolu1990, 2024-08-23:
Hi,

So, if i understand well, you want to (for example) assign a dut.enumSignal from a simulation integer value ?

KireinaHoro, 2024-08-23:
Not only enum.  So for example I have the following descriptor:

```scala
  object ProtoPacketDescType extends SpinalEnum {
    val ethernet, ip, udp, oncRpcCall = newElement()
  }

  case class ProtoPacketDescData()(implicit c: ConfigDatabase) extends Union {
    val ethernet = newElement(EthernetMetadata())
    val ip = newElement(IpMetadata())
    val udp = newElement(UdpMetadata())
    val oncRpcCall = newElement(OncRpcCallMetadata())
  }

  /**
   * [[ProtoPacketDesc]] plus type information. Used between [[RxPacketDispatch]] and [[CoreControlPlugin]].
   */
  case class TaggedProtoPacketDesc()(implicit c: ConfigDatabase) extends Bundle {
    override def clone = TaggedProtoPacketDesc()

    val ty = ProtoPacketDescType()
    val metadata = ProtoPacketDescData()
  }
```

Dolu1990, 2024-08-29:
Hi, sorry for the delay 

new SimDataPacker(TaggedProtoPacketDesc())   isn't possible yet, you will need to refer to some already existing hardware
new SimDataPacker(dut.xx.yy.taggedProtoPacketDescInstance)

```scala
    class SimDataPacker[T <: Data](data: T) {
      val layout = mutable.LinkedHashMap[Data, Int]()
      def analyse(that : Data, offset : Int): Unit = that match {
        case bt : BaseType => layout(bt) = offset
        case d : Bundle => {
          var ptr = 0
          for((_,e) <- d.elements){
            analyse(e, ptr)
            ptr += e.getBitsWidth
          }
        }
      }
      analyse(data, 0)

      // stuff some actual value into the packer
      def update(locator: T => Data, value: BigInt): Unit

      // read back part of the packed data
      def apply(locator: T => Data): BigInt
    }

```

What do you want the update and apply function to do exactly ? 
Do you want them to read write the dut hardware ? or you want them to only read write in the scala world (testbench) ?

KireinaHoro, 2024-08-29:
Hi Dolu,

I'm thinking of update and apply to only read/write in the scala world--if I need to interact with real hardware, that would mean that there's a signal with the respective type is already there.  In that case, the old `#=` and `simGet` interface should be enough.  This is more about getting a bit vector in simulation and then trying to parse it.

Dolu1990, 2024-08-29:
Ahhh i see.
So, SimDataPacker could create an internal hashmap where the key are the field, and the value would be a BigInt.
that hashmap can be populated using the hardware "data.flatten" 
But overall that is quite heavy weight, and the scope of such feature could go very deep XD
You maybe just better of typing some "pure" scala data structure instead.