SpinalHDL/SpinalHDL #1512: Inconsistent naming for nested `Composite` areas

KireinaHoro, 2024-08-22:
I'm trying to sort out why my generated Verilog sometimes have these annoying `toplevel_` prefixes for some signals, but not for others:

```verilog
  reg                 EthernetDecoder_logic_decoder_io_header_ready;
  reg                 EthernetDecoder_logic_metadata_fork_io_outputs_0_ready;
  reg                 EthernetDecoder_logic_metadata_fork_io_outputs_1_ready;
  wire                toplevel_EthernetDecoder_logic_payload_fork_io_outputs_0_throwFrameWhen_dropper_io_drop;
  wire                toplevel_EthernetDecoder_logic_payload_fork_io_outputs_1_throwFrameWhen_dropper_io_output_ready;
  wire                toplevel_EthernetDecoder_logic_payload_fork_io_outputs_1_throwFrameWhen_dropper_io_drop;
```

Here `EthernetDecoder_logic_decoder` and `EthernetDecoder_logic_payload_fork` are components and the `io_*` signals are their IO ports.  In addition, there is another component named `toplevel_EthernetDecoder_logic_payload_fork_io_outputs_0_throwFrameWhen_dropper`, derived from the enclosing `Composite` defined as `Composite(..., "throwFrameWhen")`.

After some tracking, it seems like this is due to the behaviour of `NameableByComponent` prefixing the full hierarchical path between the source component `toplevel` and the destination component, for example `EthernetDecoder_logic_payload_fork` here:

https://github.com/SpinalHDL/SpinalHDL/blob/10c04273fdf138359589408303efa4f55e371ddb/core/src/main/scala/spinal/core/Trait.scala#L240-L245

This is further accompanied by the implementation of `getPath`, resulting in a single component `toplevel` in the returned path:

https://github.com/SpinalHDL/SpinalHDL/blob/10c04273fdf138359589408303efa4f55e371ddb/core/src/main/scala/spinal/core/Trait.scala#L222-L235

Resulting in the `Composite(..., "throwFrameWhen")` being named with the `toplevel` prefix.

The problem here is that `toplevel` is implicit and breaks the nice pattern of prefix naming by `Composite`, making it difficult to select related signals at once in waveform viewers.  In extreme cases, signals between components named with this `toplevel` prefix might result in something like `toplevel_..._toplevel_...` which is messy.

Would it be reasonable if we change `getPath` here to drop the first element of the path, if it is `toplevel`:

```diff
diff --git a/core/src/main/scala/spinal/core/Trait.scala b/core/src/main/scala/spinal/core/Trait.scala
index bc9270210..f3f683880 100644
--- a/core/src/main/scala/spinal/core/Trait.scala
+++ b/core/src/main/scala/spinal/core/Trait.scala
@@ -228,10 +228,16 @@ trait NameableByComponent extends Nameable with GlobalDataUser {
       down = down.tail
       up = up.tail
     }
-    if(common != null)
+    val fullPath = if(common != null)
       (down.reverse :+ common) ++ up
     else
       down.reverse ++ up
+
+    // drop toplevel head for more consistent signal naming
+    if (fullPath.head == globalData.toplevel)
+      fullPath.tail
+    else
+      fullPath
   }

   override def getName(default: String): String = {
```

This results in (IMO) more consistent naming of related signals under the same area:

```verilog
  reg                 EthernetDecoder_logic_decoder_io_header_ready;
  reg                 EthernetDecoder_logic_metadata_fork_io_outputs_0_ready;
  reg                 EthernetDecoder_logic_metadata_fork_io_outputs_1_ready;
  wire                EthernetDecoder_logic_payload_fork_io_outputs_0_throwFrameWhen_dropper_io_drop;
  wire                EthernetDecoder_logic_payload_fork_io_outputs_1_throwFrameWhen_dropper_io_output_ready;
  wire                EthernetDecoder_logic_payload_fork_io_outputs_1_throwFrameWhen_dropper_io_drop;
```

I can submit a PR if this is desirable.  @Dolu1990

Dolu1990, 2024-08-23:
Hi,

> I'm trying to sort out why my generated Verilog sometimes have these annoying toplevel_ prefixes for some signals, but not for others:

Yes, i'm thinking that maybe we should remove that behaviour to keep things more consistant. 
I'm currently in a trip far away from home, i will be back home this week end.
I will then take a better look at it :)

Dolu1990, 2024-08-27:
Look good to me :D
PR welcome

Thanks ^^

PR #1517 (2024-08-29): Omit toplevel in component path for naming

Closes #1512
closed 2024-08-29 by Dolu1990