SpinalHDL/SpinalHDL #1514: synchronous StreamFork valid/ready are not decoupled properly

ramielkhatibPQS, 2024-08-24:
This is the [current synchronous implementation of StreamFork](https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/Stream.scala#L1056-L1058):

```scala
input.ready := outputs.map(_.ready).reduce(_ && _)
outputs.foreach(_.valid := input.valid && input.ready)
outputs.foreach(_.payload := input.payload)
```

The `output(i).valid` is dependent on `input.ready` which in turn is dependent on `input(i).ready`. That will cause a deadlock. I believe it should be changed to this:

```scala
input.ready := outputs.map(_.ready).reduce(_ && _)
outputs.foreach(o => o.valid := input.valid && outputs.filterNot(_ == o).map(_.ready).reduce(_ && _))
outputs.foreach(_.payload := input.payload)
```

This should make `output(i).valid` dependent on every other output ready signal except its own.

hy0000, 2024-08-25:
https://github.com/SpinalHDL/SpinalHDL/blob/13e7c74d873ac71df39989d8dce67831a28cc0d7/lib/src/main/scala/spinal/lib/Stream.scala#L1038-L1040
The comment had note that you should use `synchronous` carefully

ramielkhatibPQS, 2024-08-25:
Right! I read that part, but forgot about it. Is there a chance a version that doesn't dead lock be implemented. Maybe a parameter like `noDeadLock`. Same thing for `StreamJoin`.

hy0000, 2024-08-25:
Seems impossible, `synchronous` logic inevitably leads to a dependency on ready to valid. Adjust other stream logic to support asynchronous is the better way.