SpinalHDL/SpinalHDL #1534: `PackedBundle` inside `Union`?

KireinaHoro, 2024-09-11:
Following #1337, `PackedBundle` seems like a perfect candidate to define external instruction words with holes inside the definition.  However, I'm not sure how I can fit them inside a `Union`.  Consider the following with explicit ignore bits `xb*`:

```scala
case class EciWord() extends Union {
  // ...
  val lc = newElement(LclMfwd())
  val lci = newElement(LclMfwd())
  // ...
}
case class LclMfwd() extends Bundle {
  val address = EciAddress
  val xb2 = Bits(2 bits)
  val rnode = EciNodeId
  val xb1 = Bool()
  val ns = Bool()
  val dmask = EciDmask
  val hreqId = EciHreqId
  val xb3 = Bits(3 bits)
  val opcode = EciOpcode
}
```

I can try to rewrite `LclMfwd` as a `PackedBundle`:

```scala
case class LclMfwd() extends PackedBundle {
  val address = EciAddress
  skipOver(2 bits)
  val rnode = EciNodeId
  skipOver(1 bit)
  val ns = Bool()
  val dmask = EciDmask
  val hreqId = EciHreqId
  skipOver(3 bits)
  val opcode = EciOpcode
}
```

Per my understanding, the new `LclMfwd` needs to use `pack` and `unpack` methods to generate the correct bit width; simply using `LclMfwd()` will result in the `skipOver` definitions being ignored.  What's the intended way to use a `PackedBundle` inside the `Union`?

Dolu1990, 2024-09-14:
@kleinai @dokleina This issue is related to PackedBundle :)

Dolu1990, 2024-09-17:
> I could upgrade PackedBundle to overload all of the MultiData methods to implicitly use the packed form. My only concern is that it could make unnecessary logic for the unused bits.

I would say that the way to go, and then if the user could specifiy if all the skipOver should be ignored or not. something like that.

> Or maybe Union could have some special logic to handle PackedBundle?

Seems like a deep rabbit hole / unpure <3

dokleina, 2024-09-29:
I just pushed the start of a fix for this in 4fd92e64f24740eb1deab7ec852f05568eb75b41. Let me know if it works for you.

I'll be adding some additional test support for the `assignFromBits` slice form, since I don't believe any tests actually exercised the `unpack` slice form.

KireinaHoro, 2024-09-30:
@dokleina thanks for the good work!

I've tested it with the above struct definition and tried to assign like the following:

```scala
val ret = Stream(EciWord())

// generating a LCI -- refer to Table 7.9 of CCKit
ret.payload.lci.opcode  := B("00001")
ret.payload.lci.hreqId  := B(idx, EciCmdDefs.ECI_HREQID_WIDTH bits)
ret.payload.lci.dmask   := B("1111")
ret.payload.lci.ns      := True
ret.payload.lci.rnode   := B("01")
ret.payload.lci.address := addr.payload

ret.payload.assignDontCareToUnassigned()

ret.arbitrationFrom(addr)
```

Despite the `assignDontCareToUnassigned`, SpinalHDL complains that there are 6 bits unassigned (which corresponds to the three `skipOver` definitions):

```
 Error detected in phase PhaseCheck_noLatchNoOverride
********************************************************************************
********************************************************************************

LATCH DETECTED from the combinatorial signal (toplevel/bindLci_ret_payload :  Bits[64 bits]), unassigned bit mask is 11111100000000000000000000000000_00000000000000000000000000000000, defined at
    jsteward.blocks.eci.EciWord.postInitCallback(EciCmdDefs.scala:222)
    pionic.host.eci.EciInterfacePlugin$$anon$1$$anon$7.$anonfun$ret$1(EciInterfacePlugin.scala:163)
    spinal.lib.Stream.<init>(Stream.scala:57)
    spinal.lib.StreamFactory.apply(Stream.scala:36)
    spinal.lib.StreamFactory.apply(Stream.scala:41)
    pionic.host.eci.EciInterfacePlugin$$anon$1$$anon$7.<init>(EciInterfacePlugin.scala:163)
    pionic.host.eci.EciInterfacePlugin$$anon$1.$anonfun$new$13(EciInterfacePlugin.scala:162)
    pionic.host.eci.EciInterfacePlugin$$anon$1.<init>(EciInterfacePlugin.scala:162)
    pionic.host.eci.EciInterfacePlugin.$anonfun$logic$1(EciInterfacePlugin.scala:47)
    spinal.lib.misc.plugin.FiberPlugin$$anon$1.$anonfun$build$4(Fiber.scala:84)
    spinal.lib.misc.plugin.PluginHost.rework(Host.scala:29)
    spinal.lib.misc.plugin.FiberPlugin$$anon$1.$anonfun$build$2(Fiber.scala:82)
    spinal.lib.misc.plugin.FiberPlugin$$anon$1.$anonfun$build$1(Fiber.scala:79)
```

What is a bit concerning is that the `skipOver` bits are located at the end of the bundle, instead of at the correct places.  Can you check that this is actually correct?  In addition, maybe you can provide an option in the constructor of `PackedBundle` to specify, how to initialize the padding fields (e.g. assign `DontCare` or zero)? 

Thanks!

Dolu1990, 2024-10-04:
Hi,

I think you could aswell do a : 
myRecord.raw.assignDontCare before starting the assignements
SpinalHDL allows overriding don't care value without error. See : 

```scala
    val raw = Bits(32 bits)
    raw.assignDontCare()
    raw(3 downto 0) := 2
    raw(7 downto 4) := 3
```    

> ret.payload.assignDontCareToUnassigned()

That would also be a possibility, but more complicated, as it would need to check that there is no assignements in the current conditional scope (to work in all cases)

KireinaHoro, 2024-10-04:
Hi @Dolu1990, that would work yes (better to have it inside the Union?  I'm not sure).

It's a bit more hassle to assign zero though, but still could be useful--dontCare bits inside an enum would make it very difficult to look at the bundle value in simulation.

dokleina, 2024-10-05:
@KireinaHoro I already have the `pack with skipOver` and `unpack with skipOver` test cases in [SpinalSimPackedBundleTester](https://github.com/SpinalHDL/SpinalHDL/blob/0b1d1e9222755cda45793bfa2403b48cbf5c0936/tester/src/test/scala/spinal/lib/SpinalSimPackedBundleTester.scala) to test that `skipOver` does what it advertises.

I just pushed https://github.com/dokleina/SpinalHDL/commit/ab8bdef92a0134677cca9e2c9c34a9d5005c0c16. I added `setDefaultValue` to add a default value to the resulting Bits from `asBits`. Without any calls to `setDefaultValue`, the resulting Bits will be empty with `assignDontCare` applied. Not sure if this solves your problem though 😟

dokleina, 2024-10-05:
@Dolu1990 I might be able to clear some of this up if I had the equivalent of VHDL variables for range assignment in PackedBundle's `asBits`.

I'd be looking for something like this (excuse the VHDL handwaving syntax):
```vhdl
As_Bits_PackedBundle : function(packedBundle : PackedBundle)
return Bits is
     variable bits_out : Bits(packedBundle.getBitsWidth);
begin
    for field in packedBundle.fields loop
        bits_out(field.l to field.r) := field.bits
    end loop;
    return bits_out
end function;
```
But it doesn't look like Spinal can do slice assignments, since I get this error when I swap `:=` with `\=`:
```
value \= is not a member of spinal.core.Bits
  Expression does not convert to assignment because receiver is not assignable.
        packed(range) \= data.asBits.takeLow(range.size.min(data.getBitsWidth)).resize(range.size)
```

Dolu1990, 2024-10-05:
@dokleina 

```scala
var bits_out = Bits(xxx)
for(field <- packedBundle.fields) bits_out(field.r to field.l) \= field.bits
```

should be ok, need to be a var

dokleina, 2024-10-05:
@Dolu1990 I've tried it with `var` instead and it still isn't working 😞 

More complete example:
```scala
val list: List[(Range, Data)] = List (
  (0 to 5) -> B(257),
  (6 to 10) -> B(15)
)

var outBits = Bits(11 bits)
for ((r, d) <- list) {
  outBits(r) \= d.asBits
}
```
Same kind of message:
```
value \= is not a member of spinal.core.Bits
  Expression does not convert to assignment because receiver is not assignable.
      outBits(r) \= d.asBits
```

Of course this might not even fix the topic of this issue. I can break this out into a separate issue and try to track it down myself

Dolu1990, 2024-10-07:
Ahhhh right, it isn't supported for partial field, forgot that XD
overall why do you need to use \= ? why not directly := ?

dokleina, 2024-10-19:
@Dolu1990 I'd like to use `\=` so that the result of `asBits` is a purely combinatorial result of the fields within the `PackedBundle`. Basically I want `asBits` as a "view" into the bits of all of the fields