SpinalHDL/SpinalHDL #1545: Redundant Modules 'module_1', 'module_2'... Created for Different Clock Domains

kartikp4892, 2024-09-24:
**Description**: When using any components (Let's take BufferCC for example), component within different clock domains, SpinalHDL generates separate module definitions (BufferCC, BufferCC_1, etc.), even though the logic within these modules is identical. The only differences between the modules are the names of the clock and reset ports, which vary based on the clock domain signals used in each instance.

**Problem**: This behavior results in redundant component generation for the same functionality, which is inefficient and leads to multiple modules in the generated Verilog code. The issue is illustrated by the following simplified code:
```
import spinal.core._
import spinal.lib._

class PLL() extends BlackBox {
  val io = new Bundle {
    val clk_i = in Bool()
    val rst_n_i = in Bool()
    val lock_o = out Bool()
    val clk_o = out Bool()
  }

  mapClockDomain(clock=io.clk_i, reset=io.rst_n_i, resetActiveLevel=LOW)
}

class ComponentNameIssue() extends Component {

  val io = new Bundle {
    val data_in = in Bool() 
    val data_out = out Bool()
  }
  noIoPrefix()

  val pll = new PLL()

  val pllDomain =  ClockDomain.internal("pll", config = ClockDomainConfig(resetActiveLevel=LOW))
  pllDomain.clock := pll.io.clk_o
  pllDomain.reset := pll.io.lock_o

  val data_in = pllDomain on BufferCC(io.data_in)

  io.data_out := BufferCC(data_in)
}

object ComponentNameIssue {
  def main(args: Array[String]): Unit = {
    SpinalConfig().generateVerilog(new ComponentNameIssue)
  }
}

```

**Generated Verilog**: Two different BufferCC modules are generated with differing clock and reset port names:

```
module BufferCC_1 (
  input  wire          io_dataIn,
  output wire          io_dataOut,
  **input  wire          clk,
  input  wire          reset**
);

  (* async_reg = "true" *) reg                 buffers_0;
  (* async_reg = "true" *) reg                 buffers_1;

  assign io_dataOut = buffers_1;
  always @(posedge clk) begin
    buffers_0 <= io_dataIn;
    buffers_1 <= buffers_0;
  end
endmodule

module BufferCC (
  input  wire          io_dataIn,
  output wire          io_dataOut,
  **input  wire          pll_clk,
  input  wire          pll_resetn**
);

  (* async_reg = "true" *) reg                 buffers_0;
  (* async_reg = "true" *) reg                 buffers_1;

  assign io_dataOut = buffers_1;
  always @(posedge pll_clk) begin
    buffers_0 <= io_dataIn;
    buffers_1 <= buffers_0;
  end
endmodule
```

**Expected Behavior**: When a component is instantiated across different clock domains, the generated Verilog should have consistent clock and reset port names. The clock and reset signals should be mapped to the appropriate clock domain signals during instantiation without requiring separate module definitions. This would avoid redundant component generation while still maintaining proper functionality across different clock domains.

Dolu1990, 2024-09-24:
Maybe the one rule which could be added, is that if a clockdomain is pulled into a component and that clockdomain is the default one of the component => use clk/reset to name those pull ?

I implemented that here : 
https://github.com/SpinalHDL/SpinalHDL/commit/372aa664fc465df85058e81fbfed4e29e3356519

With your code example it won't work because the to clockdomain have a different configuration (reset active high / low)
But else it is ok i think.

Dolu1990, 2024-09-24:
Note it isn't turned on by default

kartikp4892, 2024-09-24:
Thanks, that was really fast. The updates are working fine, and I am closing this issue.

kartikp4892, 2024-09-27:
Reopening the issue because the below commit improperly renames the clock and reset ports of the Blackbox. The expected behavior is to rename the clk and reset signals of Components only.

https://github.com/SpinalHDL/SpinalHDL/commit/372aa664fc465df85058e81fbfed4e29e3356519

Dolu1990, 2024-09-27:
Should be good now ^^

commit ec0e4ea8 (2024-09-27): Fix #1545 prevent blackbox from having their clock renamed by the pull clockdomain phase
closed 2024-09-27 by Dolu1990