SpinalHDL/SpinalHDL #1559: Physical Number representation wrong on Scala 2.13

Readon, 2024-10-07:
Scala 2.13's f-string parser would not call formatTo function in Formattable. 
This lead following lines failed.
``` scala
assert(f"${HertzNumber(5123000)}" == "5.123 MHz")
assert(f"${HertzNumber(5123000)}%s" == "5.123 MHz")
assert(f"${HertzNumber(5123000)}%S" == "5.123MHz")
.....
``` 

Also fails the test spinal.core.PhysicalNumbers

Readon, 2024-10-07:
@andreasWallner Can you have a look at it? the test on 2.13 is also in PR #1558

andreasWallner, 2024-10-08:
I'll have a look, wondering how that ever worked then/when it broke...

Dolu1990, 2024-10-08:
@andreasWallner A toke a little look, that is realy weird, seems like string interpolation doesn't trigger in the same way between 2.13 and 2.12 XD

andreasWallner, 2024-10-08:
It's interesting since I did use that with a 2.13 version in the past...

Readon, 2024-10-09:
> @andreasWallner A toke a little look, that is realy weird, seems like string interpolation doesn't trigger in the same way between 2.13 and 2.12 XD

In 2.13 the Scala community refacted the implementation of f-interpolator.

Readon, 2024-10-09:
What does this used for? If it is really important, create a standalone custom interpolator like p-interpolator might be a good solution.

Readon, 2024-10-13:
@andreasWallner How about move to fp-interpolator? #1561

PR #1561 (2024-10-14): add fp-interpolator to avoid errors in 2.13

Closes #1559 

# Context, Motivation & Description

avoid use official f-interpolator, use fp"xxxxx". 
This will change API.

# Impact on code generation

# Checklist

- [ ] Unit tests were added
- [x] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RTD](https://github.com/SpinalHDL/SpinalDoc-RTD/issues/new?title=Document%20XXX&body=Do%20not%20merge%20until%20SpinalHDL/SpinalHDL%23XXX%20has%20not%20been%20merged)?
closed 2024-10-14 by Readon