SpinalHDL/SpinalHDL #1568: SpinalHDL's test with cocotb could be failed while using cocotb 1.9.0 and later.

Readon, 2024-10-15:
as [this CI runs](https://github.com/SpinalHDL/SpinalHDL/actions/runs/11329130739) demonstrated.

Readon, 2024-10-15:
While running InternalClockTester the error message is:
``` bash
 Traceback (most recent call last):
                                                          File "/workspaces/SpinalHDL/tester/src/test/python/spinal/InternalClockTester/InternalClockTester.py", line 21, in test1
                                                            if int(dut.io_internalClkCounter) == 2:
                                                          File "/usr/local/lib/python3.10/dist-packages/cocotb/handle.py", line 915, in __int__
                                                            return int(self.value)
                                                          File "/usr/local/lib/python3.10/dist-packages/cocotb/binary.py", line 584, in __int__
                                                            return self.integer
                                                          File "/usr/local/lib/python3.10/dist-packages/cocotb/binary.py", line 397, in integer
                                                            return self._convert_from(self._str)
                                                          File "/usr/local/lib/python3.10/dist-packages/cocotb/binary.py", line 277, in _convert_from_unsigned
                                                            return int(x.translate(_resolve_table), 2)
                                                        ValueError: invalid literal for int() with base 2: 'xxxxxxxx'
```

Dolu1990, 2024-10-21:
Hi,

I think we need to fix cocotb to a final version which won't move anymore. No new cocotb testbench is expected to be added. So it just need to be able to run with minimal maintenance.

Dolu1990, 2024-10-21:
Otherwise, all the cocotb testbench would need to be updated to support X values. So far i placed "cocotbXHack" to force cocotb handeling X values as zeroes.

Readon, 2024-10-21:
> Hi,
> 
> I think we need to fix cocotb to a final version which won't move anymore. No new cocotb testbench is expected to be added. So it just need to be able to run with minimal maintenance.

Now the version has been fixed to 1.8.1.

Readon, 2024-10-21:
> Otherwise, all the cocotb testbench would need to be updated to support X values. So far i placed "cocotbXHack" to force cocotb handeling X values as zeroes.

it seems that cocotb change some internal thing, so that cocotbXHack is not working now.

Dolu1990, 2024-10-21:
Yes, there has been quite some changes in cocotb ^^

Dolu1990, 2024-10-21:
> Now the version has been fixed to 1.8.1.

Thanks :D

Dolu1990, 2024-10-21:
Yes, let's wait, there is no hurry ^^