SpinalHDL/SpinalHDL #1570: Verilog of blackbox with inout mapped to tristate port renaming issue

louiecaulfield, 2024-10-15:
I'm trying to simulate an SPI flash with a verilog model. The blackbox definition is

```scala
private[spi] final class MX25U25645G extends BlackBox {
  val io = new Bundle {
    val SCLK = in port Bool()
    val CS   = in port Bool()
    val SI   = inout port Analog(Bool)
    val SO   = inout port Analog(Bool)
    val WP   = inout port Analog(Bool)
    val SIO3 = inout port Analog(Bool)
  }

  noIoPrefix()
  addRTLPath(flow.file(getClass(), "MX25U25645G.v").toString)
}
```

I instantiate it as follows, and connect to a TriState `m.io.data(x)`

```scala
            def inoutConnect[T <: Data](inout: T, triState: TriState[T], translate: T => T = (x : T) => x) = new Area {
                require(inout.isAnalog && inout.isInOut, "inout must be an Analog inout")
                triState.read := translate(inout)
                when(triState.writeEnable) {inout := translate(triState.write)}
            }

            val flash = new MX25U25645G
            flash.io.SCLK := m.io.clock
            flash.io.CS := !m.io.chip_select
            inoutConnect(flash.io.SI,   m.io.data(0))
            inoutConnect(flash.io.SO,   m.io.data(1))
            inoutConnect(flash.io.WP,   m.io.data(2), (x: Bool) => !x)
            inoutConnect(flash.io.SIO3, m.io.data(3), (x: Bool) => !x)
```

The generated verilog fails, as the assignment to the read signals are the port names, rather than the renamed port names `analog_wrap_i[0]`

```verilog
  MX25U25645G flash (
    .SCLK (m_io_clock      ), //i
    .CS   (flash_CS        ), //i
    .SI   (analog_wrap_0[0]), //~
    .SO   (analog_wrap_1[0]), //~
    .WP   (analog_wrap_2[0]), //~
    .SIO3 (analog_wrap_3[0])  //~
  );
  assign analog_wrap_0[0] = _zz_analog_wrap_0 ? m_io_data_write[0] : 1'bz;
  assign analog_wrap_1[0] = _zz_analog_wrap_1 ? m_io_data_write[1] : 1'bz;
  assign analog_wrap_2[0] = _zz_analog_wrap_2 ? (! m_io_data_write[2]) : 1'bz;
  assign analog_wrap_3[0] = _zz_analog_wrap_3 ? (! m_io_data_write[3]) : 1'bz;
  always @(*) begin
    _zz_analog_wrap_3 = 1'b0;
    if(m_io_data_writeEnable[3]) begin
      _zz_analog_wrap_3 = 1'b1;
    end
  end

  always @(*) begin
    _zz_analog_wrap_2 = 1'b0;
    if(m_io_data_writeEnable[2]) begin
      _zz_analog_wrap_2 = 1'b1;
    end
  end

  always @(*) begin
    _zz_analog_wrap_1 = 1'b0;
    if(m_io_data_writeEnable[1]) begin
      _zz_analog_wrap_1 = 1'b1;
    end
  end

  always @(*) begin
    _zz_analog_wrap_0 = 1'b0;
    if(m_io_data_writeEnable[0]) begin
      _zz_analog_wrap_0 = 1'b1;
    end
  end

  assign flash_CS = (! m_io_chip_select);
  always @(*) begin
    m_io_data_read[0] = SI;
    m_io_data_read[1] = SO;
    m_io_data_read[2] = (! WP);
    m_io_data_read[3] = (! SIO3);
  end
```

I believe the last `always` clause should read

```
  always @(*) begin
    m_io_data_read[0] = analog_wrap_0[0];
    m_io_data_read[1] = analog_wrap_1[0];
    m_io_data_read[2] = (! analog_wrap_2[0]);
    m_io_data_read[3] = (! analog_wrap_3[0]);
  end
```

Dolu1990, 2024-10-21:
Hi,

Right.
Hmm until this is fixed, one workaround seems the following : 
In your  component, define Analog for each blackbox analog, then connect them to the blackbox analogs.

That ensure that there is some analog "island" outside the blackbox, and make it work.

louiecaulfield, 2024-10-23:
Hi,

I'm surprised I did't try that yet, but it works. Here's the code I used, for whoever's future reference

MX25U25645G.scala
```scala filename="blackbox.scala"
import spinal.core._
import spinal.lib._

final class MX25U25645G(memInit: os.Path) extends BlackBox {
  val io = new Bundle {
    val SCLK = in port Bool()
    val RESET = in port Bool()
    val CS   = in port Bool()
    val SI   = inout port Analog(Bool)
    val SO   = inout port Analog(Bool)
    val WP   = inout port Analog(Bool)
    val SIO3 = inout port Analog(Bool)
  }

  noIoPrefix()
  addGeneric("Init_File", memInit.toString)
  addRTLPath(flow.file(getClass(), "MX25U25645G.v").toString)
}
```

The connection to my TriStates

```scala
val flash = new MX25U25645G(memInitFile)
flash.io.SI   <> flashDataIo(0).asAnalog()
flash.io.SO   <> flashDataIo(1).asAnalog()
flash.io.WP   <> flashDataIo(2).asAnalog()
flash.io.SIO3 <> flashDataIo(3).asAnalog()
```

Where `asAnalog()` is defined in a TriState pimper class

```scala

implicit class TriStatePimp[T <: Data](self: TriState[T]) {

  def asAnalog(): T = {
    val ret = self.dataType().setAsAnalog()
    when(self.writeEnable) {
      ret := self.write
    }
    self.read := ret
    ret
  }

}
```

Dolu1990, 2024-10-23:
Nice thanks ^^