SpinalHDL/SpinalHDL #1572: Failed to run simulation using verilator on Windows

du33169, 2024-10-22:
[The doc](https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Simulation/install/Verilator.html#windows) says:
> Add `bin` and `usr\bin` of MSYS2 into your windows PATH (ie : `C:\msys64\usr\bin;C:\msys64\mingw64\bin`)

I followed this instruction and got the following error when executing `verilator_bin.exe`:
```powershell
Program 'verilator_bin.exe' failed to run: The specified executable is not a valid application for this OS platform.
```

However, Verilator works fine in the MSYS2 shell.
It turns out that `C:\msys64\mingw64\bin\verilator_bin.exe` is some kinds of symbolic link with a size of around 1 KB, and cannot be properly recognized by the native Windows system.
The actual executable is located at `C:\msys64\mingw64\share\verilator\bin\verilator_bin.exe`, and it runs correctly in PowerShell.

We need to modify the PATH as follows to allow Spinal to locate Verilator successfully:
```
C:\msys64\usr\bin;C:\msys64\mingw64\share\verilator\bin;C:\msys64\mingw64\bin
```
Note that `C:\msys64\mingw64\share\verilator\bin` is placed first to ensure the correct executable is found.

However, during compilation, a new error occurred:
```powershell
[Progress] Verilator compilation started
%Error: Cannot find verilated_std.sv containing built-in std:: definitions: /mingw64/share/verilator\include\verilated_std.sv
%Error: This may be because there's no search path specified with -I<dir>.
        ... Looked in:
             /mingw64/share/verilator\include\verilated_std.sv
             /mingw64/share/verilator\include\verilated_std.sv.v
             /mingw64/share/verilator\include\verilated_std.sv.sv
             verilator\/mingw64/share/verilator\include\verilated_std.sv
             verilator\/mingw64/share/verilator\include\verilated_std.sv.v
             verilator\/mingw64/share/verilator\include\verilated_std.sv.sv
%Error: Exiting due to 2 error(s)
Exception in thread "main" java.lang.AssertionError: assertion failed: Verilator invocation failed
```
The file is present at `C:\msys64\mingw64\share\verilator\include\verilated_std.sv`.
The error seems to be related to the path handling of the Verilator compiled for MSYS2 when running in a native environment.

---
Has anyone successfully run a Verilator simulation on Windows?

Dolu1990, 2024-10-28:
I do not use it on windows anymore.
Hmmm
Should we just add the VERILATOR_ROOT tricky in the doc ?

Dolu1990, 2024-10-28:
What version of verilator did you used ? maybe it is also related ?

du33169, 2024-10-28:
> What version of verilator did you used ? maybe it is also related ?

v5.024, directly installed using pacman. I tried both mingw-w64-x86_64-verilator and mingw-w64-ucrt-x86_64-verilator. 

> Should we just add the VERILATOR_ROOT tricky in the doc ?

The two problems here seem to be primitive and related to MSYS2. Did it work when the doc was originally written? I'm not sure if they are caused by some recent changes, or if something just went wrong in my environment.

Dolu1990, 2024-10-28:
As far as i remember it worked.

Readon, 2024-10-28:
> v5.024, directly installed using pacman. I tried both mingw-w64-x86_64-verilator and mingw-w64-ucrt-x86_64-verilator.
>
>
> The two problems here seem to be primitive and related to MSYS2. Did it work [...]

Seems version 5 is not supported well if I remember right.

Dolu1990, 2024-10-29:
hmm right, better stick with 4.2xx

du33169, 2024-11-08:
Thanks! I switched to verilator v4.228 and everything works fine now.

The instructions for installing verilator [From the MinGW package manager](https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Simulation/install/Verilator.html#from-the-mingw-package-manager) are outdated, as MSYS2 no longer maintains pre-built packages for verilator v4.xx. I had to compile it from source.

> Seems version 5 is not supported well if I remember right.

Given the statement in the doc that `Ideally, the latest v4.xxx and v5.xxx is well supported`, I still hope that compatibility with verilator v5.xxx can be fixed. And v5.xxx works fine with spinalHDL on my other Linux machine.

However, there are some other problems when using v5.xxx on Windows, for example, the simulation process won't exit even after it finishes. So it may take a bit more effort than just adding the above tricks to the doc.

Readon, 2024-11-08:
This is the version that I always suggest to use all-in-one environment.

Readon, 2024-11-08:
Because of the rolling release of MSYS2, verilator 4.228 could not be found at msys2 repos now.
I have rebuild and release it at [here](https://github.com/Readon/MINGW-SpinalHDL/releases/download/0.41/mingw-w64-x86_64-verilator-4.228-1-any.pkg.tar.zst). test and feed back is welcome.

Readon, 2024-11-08:
I have repeat all your steps and find that some tester still could not pass the tester for SpinalHDL itself.

du33169, 2024-12-10:
I looked into the code and found that specifying `shCommand` to use when getting the verilator version might be a better solution. In that case, the above modification and tricks are no longer needed. 

See [the PR](https://github.com/SpinalHDL/SpinalHDL/pull/1618) for details.