SpinalHDL/SpinalHDL #1574: [Error] when generate verilog with toStreamOfFragment + addFragmentLast

facebreeze, 2024-10-23:
For following component, spinal error happens:

> Exception in thread "main" spinal.core.SpinalExit: 
> 
> 
>  Bundles must have the same final class to be assigned. Either use assignByName or assignSomeByName at 
>     spinal.lib.Stream.connectFrom(Stream.scala:297)
>     spinal.lib.Stream.$less$less(Stream.scala:100)
> 

```scala
import spinal.core._
import spinal.lib._

object MemRdCmd {
  def apply(size:Int) = new MemRdCmd(size)
}
class MemRdCmd(size:Int) extends Bundle {
  val addr = UInt(log2Up(size)bits)

  override def clone = new MemRdCmd(size)
}

case class MemRdData(size:Int, width:Int) extends MemRdCmd(size) {
  val data = Bits(width bits)
}

case class RdDataCtrl(width:Int) extends Bundle {
  val last = Bool()
  val id = UInt(width bits)
}

case class MemRdCtrl(size:Int, width:Int) extends Component {
  val io = new Bundle {
    val rd_cmd0 = slave Stream Fragment(MemRdCmd(size))
    val rd_cmd1 = slave Stream Fragment(MemRdCmd(size))
    val rd_cmd_out = master Stream Fragment(MemRdCmd(size))
    val rd_data_in = slave Stream Fragment(MemRdData(size, width))
    val rd_data_out0 = master Stream Fragment(MemRdData(size,width))
    val rd_data_out1 = master Stream Fragment(MemRdData(size,width))
  }

  val rd_cmd = StreamArbiterFactory.lowerFirst.buildOn(io.rd_cmd0,io.rd_cmd1)
  val (rd_cmd_fork0, rd_cmd_fork1) = StreamFork2(rd_cmd.io.output)
  val rd_data_ctrl_in = RdDataCtrl(rd_cmd.io.chosen.getWidth)
  rd_data_ctrl_in.last := rd_cmd.io.output.last
  rd_data_ctrl_in.id := rd_cmd.io.chosen

  val rd_data_id_queue = StreamFifo(cloneOf(rd_data_ctrl_in),4)
  rd_data_id_queue.io.push << rd_cmd_fork0.translateWith(rd_data_ctrl_in)
  val rd_data_id = rd_data_id_queue.io.pop.id
  val rd_data_last = rd_data_id_queue.io.pop.last

  val rd_data = StreamJoin(io.rd_data_in, rd_data_id_queue.io.pop)
  val rd_data_demux = StreamDemux(rd_data.translateWith(rd_data.payload._1), rd_data_id,2)
  val rd_data0 = rd_data_demux(0).toStreamOfFragment.addFragmentLast(rd_data_last) //replace the last
  val rd_data1 = rd_data_demux(1).toStreamOfFragment.addFragmentLast(rd_data_last)

  io.rd_cmd_out << rd_cmd_fork1
  io.rd_data_out0 << rd_data0
  io.rd_data_out1 << rd_data1
}
```

If connect io with `<>` instead of `<<`， spinal has error:

> Exception in thread "main" spinal.core.SpinalExit: 
> 
> 
>  Can't zip [toplevel/io_rd_data_out0_payload_fragment : MemRdData] with [toplevel/rd_data0_payload_fragment : MemRdCmd]  because they don't have the same number of elements.
> First one has :
> (addr,(toplevel/io_rd_data_out0_payload_fragment_addr : out UInt[6 bits]))
> (data,(toplevel/io_rd_data_out0_payload_fragment_data : out Bits[64 bits]))
> Seconde one has :
> (addr,(toplevel/rd_data0_payload_fragment_addr :  UInt[6 bits]))

Seems second one miss the `val data`

And the following code will generate verilog successfully.

```scala
import spinal.core._
import spinal.lib._

object MemRdCmd {
  def apply(size:Int) = new MemRdCmd(size)
}
class MemRdCmd(size:Int) extends Bundle {
  val addr = UInt(log2Up(size)bits)

  override def clone = new MemRdCmd(size)
}

case class MemRdData(size:Int, width:Int) extends MemRdCmd(size) {
  val data = Bits(width bits)
}

case class RdDataCtrl(width:Int) extends Bundle {
  val last = Bool()
  val id = UInt(width bits)
}

case class MemRdCtrl(size:Int, width:Int) extends Component {
  val io = new Bundle {
    val rd_cmd0 = slave Stream Fragment(MemRdCmd(size))
    val rd_cmd1 = slave Stream Fragment(MemRdCmd(size))
    val rd_cmd_out = master Stream Fragment(MemRdCmd(size))
    val rd_data_in = slave Stream Fragment(MemRdData(size, width))
    val rd_data_out0 = master Stream Fragment(MemRdData(size,width))
    val rd_data_out1 = master Stream Fragment(MemRdData(size,width))
  }

  val rd_cmd = StreamArbiterFactory.lowerFirst.buildOn(io.rd_cmd0,io.rd_cmd1)
  val (rd_cmd_fork0, rd_cmd_fork1) = StreamFork2(rd_cmd.io.output)
  val rd_data_ctrl_in = RdDataCtrl(rd_cmd.io.chosen.getWidth)
  rd_data_ctrl_in.last := rd_cmd.io.output.last
  rd_data_ctrl_in.id := rd_cmd.io.chosen
  val rd_data_id_queue = StreamFifo(cloneOf(rd_data_ctrl_in),4)
  rd_data_id_queue.io.push << rd_cmd_fork0.translateWith(rd_data_ctrl_in)

  val rd_data_id = rd_data_id_queue.io.pop.id
  val rd_data_last = rd_data_id_queue.io.pop.last
  val rd_data = StreamJoin(io.rd_data_in, rd_data_id_queue.io.pop)
  val rd_data_demux = StreamDemux(rd_data.translateWith(rd_data.payload._1), rd_data_id,2)

  val rd_data0 = cloneOf(rd_data_demux(0))
  val rd_data1 = cloneOf(rd_data_demux(1))
  rd_data0 arbitrationFrom rd_data_demux(0)
  rd_data0.last := rd_data_last
  rd_data0.addr := rd_data_demux(0).addr
  rd_data0.data := rd_data_demux(0).data
  rd_data1 arbitrationFrom rd_data_demux(1)
  rd_data1.last := rd_data_last
  rd_data1.addr := rd_data_demux(1).addr
  rd_data1.data := rd_data_demux(1).data

  io.rd_cmd_out << rd_cmd_fork1
  io.rd_data_out0 << rd_data0
  io.rd_data_out1 << rd_data1
}
```

Therefore, it seems that it's `toStreamOfFragment` or `addFragmentLast` that cause the issue.

Dolu1990, 2024-10-23:
Hi,

That was a tricky one XD

So you need to override the clone function all the way : 

```scala
case class MemRdData(size:Int, width:Int) extends MemRdCmd(size) {
  val data = Bits(width bits)

  override def clone = new MemRdData(size, width)
}
``` 
Else, when SpinalHDL will try to create type copy of MemRdData, it would get the MemRdCmd (as you overrided it already in MemRdCmd :)

closed 2024-10-23 by Dolu1990

facebreeze, 2024-10-24:
Hi, Dolu,
     Thanks a lot.  I make a new project and override clone function in `MemRdData`,  the problem get solved.  

     However, I have been tried override clone function in `MemRdData`， and the problem still exitst,  therefore  I opened this issue, Looks like something wrong happeded during compile.

facebreeze, 2024-10-24:
And BTW, is it okay to add a `replaceLast` method for Fragment instead of  `bundle.toStreamOfFragement.addFragmentLast`  ?

Dolu1990, 2024-10-24:
> replaceLast

Sure, PR welcome ^^