SpinalHDL/SpinalHDL #1579: DfiControllerTester randomly fail

Dolu1990, 2024-10-28:
Hi @liyaohou

It seems that DfiControllerTester sometime fail its tests.
Here is a easy way to reproduce in DfiControllerTester.scala : 

```scala
  test("DfiControllerTester") {
    for(i <- 0 until 100)  <- add this line
    SimConfig ....
    .doSimUntilVoid (seed = i){ dut =>     <- add (seed = i)
```

For instance, for me it fail at seed 11.

Can you reproduce the issue on yourside aswell ?

Readon, 2024-10-28:
I think this tester should be refacted with BmbMasterAgent and DfiSlaveAgent (need to be newly added).
How about comment the failed tester first?

Dolu1990, 2024-10-29:
disabled.

Readon, 2024-10-29:
I have tried the tester, the tester fail when I fixed the seed to 1111240514.
However, when I use withFstWave in the tester, it would magically pass the test.