SpinalHDL/SpinalHDL #1582: Empty module name in generated verilog

petrmikheev, 2024-10-29:
In instance of [StreamFifoCC](https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/Stream.scala#L1570). SpinalHDL generated invalid verilog:

```verilog
  (* keep_hierarchy = "TRUE" *) BufferCC popToPushGray_buffercc (
    .io_dataIn  (popToPushGray[3:0]                    ), //i
    .io_dataOut (popToPushGray_buffercc_io_dataOut[3:0]), //o
    .clk_cpu    (clk_cpu                               ), //i
    .reset_cpu  (reset_cpu                             )  //i
  );
  (* keep_hierarchy = "TRUE" *) BufferCC_1  (   // <===== module name is empty
    .io_dataIn  (_zz_3      ), //i
    .io_dataOut (_io_dataOut), //o
    .clk        (clk        ), //i
    .reset_cpu  (reset_cpu  )  //i
  );
  (* keep_hierarchy = "TRUE" *) BufferCC_2 pushToPopGray_buffercc (
    .io_dataIn  (pushToPopGray[3:0]                    ), //i
    .io_dataOut (pushToPopGray_buffercc_io_dataOut[3:0]), //o
    .clk        (clk                                   ), //i
    ._zz_1      (_zz_4                                 )  //i
  );
```

SpinalHDL version: commit c796d9fc7.

petrmikheev, 2024-10-29:
It happened when I accidentally moved `ClockDomainStack.set` from the beginning of a module to the middle, so different parts turned out to be in different clock areas and SpinalHDL autogenerated a huge number of CC for every bus. So it was a bug on my side and in normal workflow everything works fine. But anyway, an empty module name is suspicious.

Dolu1990, 2024-10-30:
Hi,

I never had encounter this issue. 
Hmm, would need an example to reproduce.

If by any chance you can provide us a code to reproduce, let me know ^^

petrmikheev, 2024-10-30:
> If by any chance you can provide us a code to reproduce, let me know ^^

Okay, seems it is time to make my repo public... Hope it will not be too difficult to build.

https://github.com/petrmikheev/endeavour

```bash
$ git clone --recurse-submodules https://github.com/petrmikheev/endeavour.git
$ git checkout reproduce_spinalhdl_bug
$ cd rtl/spinal && sbt 'runMain endeavour.EndeavourSoc'
$ cd ../testbench && make
```

It results in
```
../spinal/EndeavourSoc.v:38455: syntax error
../spinal/EndeavourSoc.v:38454: error: Invalid module instantiation
```

petrmikheev, 2024-11-05:
Great!

closed 2024-11-05 by petrmikheev