SpinalHDL/SpinalHDL #1583: simPublic() doesn't work when using XSim backend

Pyromuffin, 2024-10-30:
I might be the only person in the world right now using the xsim backend, but we need to use it for simulating Vivado IP :)

This very simple testbench fails with xsim but not in verilator:

``` 
class Test extends Component {

    val counter = Counter(10 bits)
    counter.increment()
    counter.value.simPublic()

  }
```

and

```
config.compile(new Test()).doSim { dut =>

    dut.clockDomain.forkStimulus(period = 100)

    dut.clockDomain.waitSampling(100)
    println(dut.counter.value.toInt)
    simSuccess()
}
```

xsim result:
```
UNACCESSIBLE SIGNAL : (toplevel/counter_value :  UInt[10 bits]) isn't accessible during the simulation.
- To fix it, call simPublic() on it during the elaboration.
```

verilator result

```
[Progress] Start Test test simulation with seed 1331964570
99
[Done] Simulation done in 16.743 ms
```

I'm not sure why this is happening but I don't think it is intended.

Thanks!

Readon, 2024-10-30:
It seems that XSimBackend do not have SimPublic processing code now.
The GHDL, iverilog and VCS backend use Vpi by [code](https://github.com/SpinalHDL/SpinalHDL/blob/70750fc0e74703cd821cf3a31bcfea7eb38ddf79/core/src/main/scala/spinal/core/sim/SimBootstraps.scala#L393).
The verilator is done by SimVerilatorPhase.

However, XSimBackend seems do not have one, (to my knowledge).

Pyromuffin, 2024-10-31:
It appears that the fix is not as simple as adding those simPublic signals to the signal collector. I tried doing that in a test build and it's causing horrible JVM access violations in spinal.sim.xsi.JNIXSIIfaceJNI.XSIIface_get_signal_handle. Continuing to investigate.

Dolu1990, 2025-04-06:
Notice [onlyStdLogicVectorAtTopLevelIo](https://github.com/SpinalHDL/SpinalTemplateSbt/commit/976ceca3a31ebc0805f978eb08af749b575c9060)