SpinalHDL/SpinalHDL #1591: Mem init nested Bundle asBits not considered as literal

oletf, 2024-11-06:
I'm having this specific case where I'm initializing a Mem from Bundles for which all elements (even nested) has been assigned by literals.

```scala
    case class OtherBundle() extends Bundle {
      val b1 = Bits(4 bits)
      val b2 = Bits(4 bits)
    }
    case class TestBundle() extends Bundle {
      val b     = Bool()
      val bs    = Bits(3 bits)
      val other = Bits(8 bits)
    }
    val other = OtherBundle()
    other.b1 := B"1010"
    other.b2 := B"0101"
    val testBundle = TestBundle()
    testBundle.b     := True
    testBundle.bs    := B"101"
    testBundle.other := other.asBits
    val memTest = Mem(Seq(testBundle))
```

here I get the 
> ROM initial value should be provided from full literals value

error, it seems like apparently

oletf, 2024-11-06:
indexing the `Seq` without going through the `Mem` works without problem though !

Dolu1990, 2024-11-07:
Hi,

Right, Mem Rom only support direct literals values. SpinalHDL will no try to compute the value asBits. (by design)
There is no workaround currently, the user need to provide a value explicitly.

I guess it would not realy be too hard to compute the asBits value at elaboration time. But that can lead to a deep rabbit hole .

oletf, 2024-11-07:
alright thanks !
will stick to the `Mem`less `Seq` for now, 
and will figure a way around next time I'll need a `Mem` for this.

closed 2024-11-07 by oletf