SpinalHDL/SpinalHDL #1597: Assigning scala decimal to AFix is incorrect

dingjr7, 2024-11-19:
As the doc claimed,
```
AFix.S(2 exp, -2 exp) can represent: -2.0, -1.75, -1.5, -1.25, -1, -0.75, -0.5, -0.25, 0, 0.25, 0.5, 0.75, 1, 1.25, 1.5, 1.75
```
However, the following codes fails to compile
```
val half_litera = AFix.S(2 exp, -2 exp)
half_litera := -0.5
```
with the error,
```
Exception in thread "main" java.lang.Exception: literal value is negative and cannot be represented
	at spinal.core.internals.BitsLiteral$.apply(Expression.scala:2314)
	at spinal.core.internals.BitsLiteral$.apply(Expression.scala:2327)
	at spinal.core.B$.$anonfun$getFactory$1(Literal.scala:124)
	at spinal.core.B$.$anonfun$getFactory$1$adapted(Literal.scala:124)
	at spinal.core.BitVectorLiteralFactory.apply(Literal.scala:37)
	at spinal.core.package$.BigIntToBits(core.scala:283)
	at spinal.core.AFix.$colon$eq(AFix.scala:1202)
```

By replacing AFix.S with SFix, the above code compiles.

Dolu1990, 2024-11-21:
Hi, 
I just tried in SpinalHDL dev, your code sample seems to work fine in it.
Which version of SpinalHDL are you using ?

dingjr7, 2024-11-23:
It's 1.10.2a. My colleague suggested that it has been fixed in https://github.com/SpinalHDL/SpinalHDL/pull/1547