SpinalHDL/SpinalHDL #1598: Issue with `.withFstWave` Output in GTKWave for Large Simulation Time (SpinalHDL 1.10.2a)

szNightFury, 2024-11-21:
#### Description  
When running a simulation with a large time duration (~500k seconds), the waveform generated using the `.withFstWave` option cannot be opened in GTKWave. Attempting to load the `.fst` file in GTKWave results in the following error:  

```
GTKWave Analyzer v3.4.0 (w)1999-2022 BSI

GTKWAVE | Could not initialize 'D:\xxx\wave.fst'.
GTKWAVE | File load failure, new tab not created.
```

However:  
- Using `.withVcdWave` works fine.  
- Converting the generated `.vcd` file to `.fst` using GTKWave's `-o` option also allows the `.fst` to load correctly in GTKWave.  

#### Environment  
- SpinalHDL version: 1.10.2a  
- GTKWave versions: Tested with multiple versions, all producing the same result.  

#### Questions  
1. Is this issue due to SpinalHDL's `.withFstWave` implementation?  
2. Could this be a compatibility issue or limitation in GTKWave?  
3. Why is the simulation time unit displayed as sec? Even after explicitly setting .withTimeSpec, it seems to have no effect. Is there any problem with the timescale configuration in SpinalHDL?

Dolu1990, 2024-11-21:
Hi,

I tested with : 

```scala
  SimConfig.withFstWave.compile(new Component{
    val x = in UInt(16 bits)
  }).doSim{dut =>
    for(i <- 0 until 1000){
      sleep(1000 sec)
      dut.x #= i
    }
  }
```

It seems to work fine for me. 
- Verilator 4.218 2022-01-17 rev v4.21
- GTKWave Analyzer v3.3.116
- SpinalHDL dev

szNightFury, 2024-11-25:
> Hi,
>
> I tested with :
>
> ```scala
>   SimConfig.withFstWave.compile(new Component{
>     val x = in UInt(16 bits)
>   }).doSim{dut =>
>     for(i <- 0 until 1000){
>       sleep(1000 sec)
>       dut.x #= i [...]

Hi @Dolu1990,

Thank you for your detailed response and clarification. After further investigation, I have some updates to share:

1. Your test case is indeed correct and works as expected.
2. Through step-by-step module decomposition, I’ve narrowed down the issue. It occurs when using Xilinx's DSP48E2.v module with cascade=256 and a test length of 40960. Under these conditions, gtkwave encounters a "load failure" while opening the .fst file.
3. I also tested the .fst file on a Linux system, and the "segmentation fault" happens when attempting to read it. However, the module works perfectly on Linux for waveform generation, and I can view the waveforms without issues (with identical versions of SpinalHDL, Verilator, and gtkwave). Transferring the .fst file generated on Linux to a Windows system also works fine.
4. After comparing results with a fixed seed and always-valid inputs (ensuring identical simulation times), I noticed that the waveform files generated on Windows are smaller than those on Linux (~2.2GB vs. ~2.5GB, though I may not recall the exact sizes).

Overall, this seems like a strange issue. It could potentially be related to the Windows version of the tools, or it might be due to a misconfiguration on my side. That said, I’ve since switched my system to Ubuntu, where everything works perfectly.

Thanks again for your excellent work on SpinalHDL and for helping me with this issue. Since everything is now running smoothly on my end, feel free to close this issue.

closed 2024-11-26 by Dolu1990