SpinalHDL/SpinalHDL #1601: getRtlPath crashes on clock signal

distributed, 2024-11-26:
Calling `getRtlPath` for a clock signal fails while it works for other signals.

Is it wrong to use `getRtlPath` on clock signals? I assumed it would work like for all other signals.

Here is a reproduction case:

```scala
package repro

import spinal.core._

class GetRtlPathCrash extends Component {
  val inp = in(Bool())
  val outp = out(Bool()) setAsReg()
  outp := inp

  println(s"outp ${outp}")
  println(s"outp rtl name: ${outp.getRtlPath()}")
  println(s"clk ${clockDomain.clock}")
  println(s"clk rtl name: ${clockDomain.clock.getRtlPath()}")
}

object GetRtlPathCrash {
  def main(args: Array[String]): Unit = {
    SpinalVerilog(new GetRtlPathCrash())
  }
}
```

Dolu1990, 2024-11-26:
Hi,
It is in part to be expected.
clockDomain.clock doesn't refer to the clock in the current component, but instead anyware it came from.
clockDomain.readClockWire.getRtlPath would work

getRtlPath seems to crash if the given signal is defined in void (outside the toplevel)

Dolu1990, 2024-11-26:
(the default clock domain   clock signals is defined in void by SpinalHDL, by design)

distributed, 2024-11-26:
Thanks, i suspected it may be something in this direction. Does this property apply to all `pull()`ed signals, then?

I stumbled across this because I want to write out a list of signals for GTKWave, but specified in Scala (still, mostly works :)). Is there a way to deduce that I need to go via e.g. `readClockWire`?

> (the default clock domain clock signals is defined in void by SpinalHDL, by design)

Meaning I can't rely on it being present, correct?

Dolu1990, 2024-11-26:
> Meaning I can't rely on it being present, correct?

For all signals of the clockdomain, durring elaboration time, the signals aren't yet pulled through.

What you can do is to use the elaborated component : 

```scala
val  report = SpinalVerilog(new MyComponent())
report.toplevel // that guy will have all the final signals named and pulled through
```

distributed, 2024-11-26:
Like this? It still crashes for me

```scala
package repro

import spinal.core._

// filed as https://github.com/SpinalHDL/SpinalHDL/issues/1601
class GetRtlPathCrash extends Component {
  val inp = in(Bool())
  val outp = out(Bool()) setAsReg()
  outp := inp
clockDomain.readClockWire
  println(s"outp ${outp}")
  println(s"outp rtl name: ${outp.getRtlPath()}")
  println(s"outp components: ${outp.getComponents()}")
  println(s"clk ${clockDomain.clock}")
  println(s"clk components ${clockDomain.clock.getComponents()}")
  //println(s"clk rtl name: ${clockDomain.clock.getRtlPath()}")
}

object GetRtlPathCrash {
  def main(args: Array[String]): Unit = {
    val report = SpinalVerilog(new GetRtlPathCrash())
    println(s"clk rtl name: ${report.toplevel.clockDomain.clock.getRtlPath()}")
  }
}
```

I guess one option for me would be to check the `.getComponents().length` and if it is zero, I have to give the signal some kind of special treatment.

Dolu1990, 2024-11-26:
More like this : 

```scala
class GetRtlPathCrash extends Component {
  val inp = in(Bool())
  val outp = out(Bool()) setAsReg()
  outp := inp
  println(s"outp ${outp}")
  println(s"outp rtl name: ${outp.getRtlPath()}")
  println(s"outp components: ${outp.getComponents()}")
  println(s"clk ${clockDomain.readClockWire}")
  println(s"clk components ${clockDomain.readClockWire.getComponents()}")
  //println(s"clk rtl name: ${clockDomain.clock.getRtlPath()}")
}

object GetRtlPathCrash {
  def main(args: Array[String]): Unit = {
    val report = SpinalVerilog(new GetRtlPathCrash())
    println(s"clk rtl name: ${report.toplevel.rework(report.toplevel.clockDomain.readClockWire.getRtlPath())}")
    report.toplevel.dslBody.walkDeclarations{
      case bt : BaseType => println(bt.getRtlPath())
      case _ =>
    }
  }
}
```

Dolu1990, 2024-11-26:
readClockWire is sensitive in which context you call it. so, report.toplevel.rework(xxx) to put yourself back into that context ^^

distributed, 2024-11-28:
Thanks, works like a charm. I'll close the issue so it won't linger around :)

closed 2024-11-28 by distributed