SpinalHDL/SpinalHDL #1602: Strange behavior of SteamMux

facebreeze, 2024-11-27:
Hi, All,
    What's the usage of the third auxiliary constructors in StreamMux ?
    In my understanding, two or more `Stream` can be muxed by a `Stream` whose `payload` is the `selector`. However, the method **`createSelector`** seems return a `Stream` without handshake with any other stream, and the **`select`** is registered, thus it will malfunction when used like this. 
    So the problem is am I understood in a wrong way or it's just a bug ?

```scala

object StreamMux {
  def apply[T <: Data](select: UInt, inputs: Seq[Stream[T]]): Stream[T] = {
    val vec = Vec(inputs)
    StreamMux(select, vec)
  }

  def apply[T <: Data](select: UInt, inputs: Vec[Stream[T]]): Stream[T] = {
    val c = new StreamMux(inputs(0).payload, inputs.length)
    (c.io.inputs, inputs).zipped.foreach(_ << _)
    c.io.select := select
    c.io.output
  }

  def apply[T <: Data](select: Stream[UInt], inputs: Vec[Stream[T]]): Stream[T] = {
    val c = new StreamMux(inputs(0).payload, inputs.length)
    (c.io.inputs, inputs).zipped.foreach(_ << _)
    select >> c.io.createSelector()
    c.io.output
  }
}
class StreamMux[T <: Data](dataType: T, portCount: Int) extends Component {
  val io = new Bundle {
    val select = in UInt (log2Up(portCount) bit)
    val inputs = Vec(slave Stream (dataType), portCount)
    val output = master Stream (dataType)
    def createSelector(): Stream[UInt] = new Composite(this, "selector") {
      val stream = Stream(cloneOf(select))
      val reg = stream.haltWhen(output.isStall).toReg(U(0))
      select := reg
    }.stream
  }
  for ((input, index) <- io.inputs.zipWithIndex) {
    input.ready := io.select === index && io.output.ready
  }
  io.output.valid := io.inputs(io.select).valid
  io.output.payload := io.inputs(io.select).payload
}
```

Readon, 2024-11-27:
Hi, may be you can have a look at #1308 for the usage.
The function create a stream need to be drived from external, it guarantees that `select` only be changed until output is not stalled, to keep the valid and payload stable while valid is asserted.
Some protocol requires stream act like that, such as AXI4.

Btw, the createSelector have been renamed to createStreamRegSelect in dev.

facebreeze, 2024-11-28:
Thanks. the method "joinSel" is what I wanted.

Readon, 2024-11-28:
Nice, how about close this issue?

closed 2024-11-29 by facebreeze