SpinalHDL/SpinalHDL #1603: SpinalHDL generate (verilog always syntax)

vivado2019, 2024-11-28:
SpinalHDL is the best hardware description language I have ever encountered, despite having written Verilog and VHDL for many years. Recently, while learning SpinalHDL, I encountered an issue: 
how to ensure that the generated Verilog code has only one always block corresponding to each signal. 
This would make the generated Verilog code much easier to maintain and read. I look forward to your reply

Dolu1990, 2024-11-28:
Thanks ^^ 

> how to ensure that the generated Verilog code has only one always block corresponding to each signal.

I'm not sure to understand.
Currently, the default SpinalHDL behaviour is : 
- Every combinatorial signal get its own always block
- Every register which use the same clockdomain have the same always block (excepted if they use async reset, there it is splited in two always block)

You would prefer that all the combinatorial signal would use the same always block ?

vivado2019, 2024-11-28:
Thanks for your reply.

For example:

This is source code:

class StateDemo_1 extends Component {
	val io = new Bundle {
		val result = out Bool()
	}

	// 独立的计数器
	val cnt = Reg(UInt(8 bits)) init (0)

	// 定义状态机
	val fsm = new StateMachine {
		val IDLE, STATE_A, STATE_B, STATE_C = new State
		setEntry(IDLE)

		IDLE
			.whenIsActive(goto(STATE_A))

		STATE_A
			.whenIsActive(goto(STATE_B))

		STATE_B
			.whenIsActive {
				when(cnt === 4)(goto(STATE_C))
					.otherwise(goto(STATE_B))
			}

		STATE_C
			.whenIsActive(goto(STATE_B))
	}

	when(fsm.isActive(fsm.STATE_B)) {
		cnt := cnt + 1
	}.otherwise {
		cnt := 0
	}

	// 输出结果
	io.result := fsm.isActive(fsm.STATE_B) && (cnt === 4)
}

This is a part of verilog by spinalHDL generated:

  always @(posedge clk or posedge reset) begin
    if(reset) begin
      cnt <= 8'h0;
      fsm_stateReg <= `fsm_enumDefinition_binary_sequential_fsm_BOOT;
    end else begin
      if(when_StateDemo_1_l33) begin
        cnt <= (cnt + 8'h01);
      end else begin
        cnt <= 8'h0;
      end
      fsm_stateReg <= fsm_stateNext;
    end
  end

however, I hope it can be like this:
  always @(posedge clk or posedge reset) begin
    if(reset) begin
		cnt <= 8'h0;
	else if(when_StateDemo_1_l33) begin
		cnt <= (cnt + 8'h01);
	end else begin
		cnt <= 8'h0;
	end
  end
  
   always @(posedge clk or posedge reset) begin
    if(reset) begin
      fsm_stateReg <= `fsm_enumDefinition_binary_sequential_fsm_BOOT;
    end else begin
      fsm_stateReg <= fsm_stateNext;
    end
  end

Thank you

Dolu1990, 2024-11-28:
Added via SpinalConfig(mergeSyncProcess=false)