SpinalHDL/SpinalHDL #1605: currentTestPath and Simulation Workdir Structure

distributed, 2024-11-29:
In a simulation, which by default uses Verilator, I have the following line: 

```scala
val writer = new BufferedWriter(new FileWriter(Paths.get(currentTestPath, filename).toString))
```

This works fine. However if I change to simulate `.withIverilog` the simulation crashes as follows:

```
[info] - triggerbank *** FAILED ***
[info]   java.io.FileNotFoundException: ./simWorkspace/TriggerBank/triggerbank/ctx.tcl (No such file or directory)
[info]   at java.base/java.io.FileOutputStream.open0(Native Method)
[info]   at java.base/java.io.FileOutputStream.open(FileOutputStream.java:295)
[info]   at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:236)
[info]   at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:124)
[info]   at java.base/java.io.FileWriter.<init>(FileWriter.java:67)
[info]   at simutils.WriteSimContextFile$.apply(WriteSimContextFile.scala:11)
[info]   at oscar.imxctrl.TriggerBankSuite.$anonfun$new$2(TriggerBankSuite.scala:55)
[info]   at spinal.core.sim.SimCompiled.$anonfun$doSimApi$1(SimBootstraps.scala:614)
[info]   at spinal.sim.SimThread.$anonfun$jvmThread$1(SimThread.scala:90)
[info]   at spinal.sim.JvmThread.run(SimManager.scala:51)
```

It seems to me that directory structure is different between the simulators. Here it is for Verilator:

```
$ tree simWorkspace/TriggerBank/
simWorkspace/TriggerBank/
├── rtl
│   └── TriggerBank.v
├── triggerbank
│   ├── ctx.tcl
│   └── wave.vcd
```

And here for Iverilator:

```
$ tree simWorkspace/TriggerBank/
simWorkspace/TriggerBank/
├── rtl
│   ├── __simulation_def.v
│   └── TriggerBank.v
├── shmem_name
├── test.vcd
└── TriggerBank.vvp
```

Dolu1990, 2024-11-29:
Hi,

Right, the other simulation backend would need to be updated to follow the same layout

distributed, 2024-12-02:
I've tried my hand at this, albeit not successfully. 

In `SimBootstraps.scala` the Verilator backend (which I've used so far) passes in a `vcdPath` that contains a placeholder `$TEST`, e.g. `vcdPath = "./simWorkspace/ApbMasterMuxBench/$TEST"`. Obviously the Verilator backend has to be aware of this, as it can't use this string verbatim, and in `instantiate(name: String, seed: Int)` it performs a substitution. This is how I understand that the string works out to be e.g. `./simWorkspace/ApbMasterMuxBench/testname` with the `wave.vcd` being tacked on later.

In the family of VPI backends the `wavePath` is hard coded to `test.vcd` in the `SpinalVpiBackend`. E.g. the IVerilog implementation just writes out a dump instruction to the simulator as ` ${if (hasWave) "$dumpfile(\"" + wavePath + "\");" else ""}`. Now the `vcdPath` seems to be interpreted not as a path with placeholders but as a file name.

Also it seems that the test case name (and probably the seed as well!) need to be passed through the IVerilog/VPI backend `instantiate` calls as they are already for e.g. the Verilator one.

I'm a bit stumped for now and left with a bunch of questions.

- Is this interpretation even correct? :upside_down_face: 
- Should the layout for all the backends be like the Verilator one, or something else?
- Should the backends agree on what kind of string the vcdPath is, e.g. "a path with placeholders onto which the actual file name including extension is suffixed?"
- `currentTestPath()` seems to follow the "placeholder approach" with `$TEST`. Should I rather change the backends to also follow this approach or should the backends declare where they put their wave files in an object, ready to be picked up by `currentTestPath()`.

I'm aware that some of these overlap a bit and was hoping that you could shed some light on a sensible direction.

Dolu1990, 2024-12-02:
> Is this interpretation even correct? 

Yes

>  ${if (hasWave) "$dumpfile(\"" + wavePath + "\");" else ""}

That is kinda the pain there, would need to find a way to programaticaly change the location of the wave file (instead of using verilog)

I asked our god and saviour chat-gpt.
He told me : 
```verilog
module testbench;
    reg clk;
    reg reset;
    string vcd_path;

    initial begin
        // Default VCD file name
        vcd_path = "output.vcd";
        
        // Check for command-line argument and update the VCD file path
        if ($value$plusargs("vcdfile=%s", vcd_path)) begin
            $display("VCD file set to: %s", vcd_path);
        end else begin
            $display("Using default VCD file: %s", vcd_path);
        end
        
        // Specify the VCD file path
        $dumpfile(vcd_path);
        $dumpvars(0, testbench);
        
        // Simulation duration control
        #100 $finish;
    end

    // Clock generation
    always #5 clk = ~clk;

    // Other testbench code...
endmodule
```

```shell
iverilog -o testbench.vvp testbench.v
vvp testbench.vvp +vcdfile=path/to/desired.vcd
```

Would do the trick XD

> Should the layout for all the backends be like the Verilator one, or something else?

Yes

> Should the backends agree on what kind of string the vcdPath is, e.g. "a path with placeholders onto which the actual file name including extension is suffixed?"

Yes

> currentTestPath() seems to follow the "placeholder approach" with $TEST. Should I rather change the backends to also follow this approach or should the backends declare where they put their wave files in an object, ready to be picked up by currentTestPath().

Hmm, i would say, the backend should follow this approach

distributed, 2024-12-02:
Thanks! Would putting the logic into `analyzeRTL` be an option? Or is `analyzeRTL` only called once per Simulator compilation but not once per test run, hence why you would like the verilog file to read the wave file path from it's arguments?

Dolu1990, 2024-12-02:
analyzeRTL is only called once.
Instead i think it should be in the runSimulation function. vpp is invoked there on each new simulation run ^^