SpinalHDL/SpinalHDL #1606: Verilator simulation should not exit if `severity=WARNING`

goekce, 2024-11-30:
```scala
case class Dut() extends Component {
  val io = new Bundle {
    val i = in Bool()
    val o = out Bool()
  }
  io.o := io.i
}
object Test extends App {
  Config.sim.compile(Dut()).doSim { dut =>
    dut.clockDomain.forkStimulus(period = 10)
      dut.clockDomain.waitRisingEdge()
      assert(dut.io.i.toBoolean != dut.io.o.toBoolean, WARNING)
      dut.clockDomain.waitRisingEdge()
      dut.clockDomain.waitRisingEdge()
  }
}
```

```
sbt "runMain Test"
```

Output

```
...
[info] [Progress] Verilator compilation started
[info] [info] Found cached verilator binaries
[info] [Progress] Verilator compilation done in 483.369 ms
[info] [Progress] Start Dut test simulation with seed 469730164
[info] [Error] Simulation failed at time=170
[error] Exception in thread "main" java.lang.AssertionError: assertion failed: spinal.core.WARNING$@e394c64
...
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
...
```
The simulation exits immediately at 170. I would expect the simulation to continue until the simulation code's end.

[According to the docs](https://github.com/SpinalHDL/SpinalDoc-RTD/blob/485f6b3c7d2e8e5dd472ddf825279f765c013476/source/SpinalHDL/Other%20language%20features/assertion.rst?plain=1#L17-L24):

```
* - NOTE
     - Used to report an informative message
   * - WARNING
     - Used to report an unusual case
   * - ERROR
     - Used to report an situation that should not happen
   * - FAILURE
     - Used to report a fatal situation and close the simulation
```

I understand that only `FAILURE` should close the simulation, not the other severities. Currently only `INFO` does not exit the simulation. Is this a bug or feature?

Readon, 2024-12-02:
To my knowledge, the assert would raise an error while the equation test failed.
If you just want to log the inequality, do not use assert, may be log utility is a good alternative.

goekce, 2024-12-02:
I believe there is a misunderstanding. In my experience, the assertions are here to check the sanity (at different severity levels like note, warning etc). Log utility is much more general. What do you think?

Dolu1990, 2024-12-05:
Hi,

I think the initial issue is a miss use of the scala runtime time assert.
assert(false, WARNING) => final def assert(assertion: Boolean, message: => Any)(implicit loc: Location) = scala.Predef.assert(assertion,message)

the WARNING here is used as a message, not as a severity level.
Scala runtime time assert (Boolean condition) do not support severity level
Only elaboration time assert does (Bool condition) via for instance def assert(assertion: Bool, severity: AssertNodeSeverity)(implicit loc: Location) 

But this is not for simulation

goekce, 2024-12-05:
So this issue could be fixed if we don't misuse Scala runtime assert but implement something like the elaboration assert, right?

Dolu1990, 2024-12-09:
Yes right. just that it is kinsa scarry to do, as Scala already provide the assert(Boolean, message), so not sure it is a good idea.

Else, there is SpinalWarning object already, but it isn't realy the same use case either XD

goekce, 2024-12-09:
I am unfamiliar with Scala, so I would like to trust experts.
If you think this is not easy to fix, I am okay with closing it to make room for other issues.

goekce, 2024-12-10:
True. This should be the least we do 🙂

Dolu1990, 2024-12-10:
Done

closed 2024-12-10 by Dolu1990