SpinalHDL/SpinalHDL #1612: Seeking a handy way to test dev code

facebreeze, 2024-12-08:
Hi, All,
     Is there a handy way to test dev code on **Windows10** ? I can successfully execute `sbt publishLocal` command to generate corresponding jar file in local ivy repository. However, I failed to import the local ivy repository in a new scala project. sbt won't find the specified spinalHDL dependency.
     The published local ivy repo resides:

[image]

And the build.sbt like this:

[image]

**sbt errors:**

[error] stack trace is suppressed; run 'last update' for the full output
[error] stack trace is suppressed; run 'last ssExtractDependencies' for the full output
[error] (update) sbt.librarymanagement.ResolveException: Error downloading com.github.spinalhdl:spinal-core_2.12:dev
[error]   Not found
[error]   Not found
[error]   not found: C:\Users\Starx\.ivy2\local\com.github.spinalhdl\spinal-core_2.12\dev\ivys\ivy.xml
[error]   not found: https://repo1.maven.org/maven2/com/github/spinalhdl/spinal-core_2.12/dev/spinal-core_2.12-dev.pom
[error] Error downloading com.github.spinalhdl:spinal-lib_2.12:dev

Dolu1990, 2024-12-09:
Hi,

Hmm that is weird.
Can you send your full project (the one with your build.sbt)
So i can try on linux, to be sure it is a windows/environnement related issue)

facebreeze, 2024-12-10:
Hi, Dolu,
     The attached `demo1.zip` is my full project.  I can successfully get spinalHDL 1.10.1 or 1.10.2 with this project, but can not get **dev** package which has been published in local.

My local ivy repo:
```
c:\Users\WangD\.ivy2\local/
└── com.github.spinalhdl
    ├── spinalhdl-core_2.12
    │   └── dev
    │       ├── docs
    │       │   ├── spinalhdl-core_2.12-javadoc.jar
    │       │   ├── spinalhdl-core_2.12-javadoc.jar.md5
    │       │   └── spinalhdl-core_2.12-javadoc.jar.sha1
    │       ├── ivys
    │       │   ├── ivy.xml
    │       │   ├── ivy.xml.md5
    │       │   └── ivy.xml.sha1
    │       ├── jars
    │       │   ├── spinalhdl-core_2.12.jar
    │       │   ├── spinalhdl-core_2.12.jar.md5
    │       │   └── spinalhdl-core_2.12.jar.sha1
    │       ├── poms
    │       │   ├── spinalhdl-core_2.12.pom
    │       │   ├── spinalhdl-core_2.12.pom.md5
    │       │   └── spinalhdl-core_2.12.pom.sha1
    │       └── srcs
    │           ├── spinalhdl-core_2.12-sources.jar
    │           ├── spinalhdl-core_2.12-sources.jar.md5
    │           └── spinalhdl-core_2.12-sources.jar.sha1
    ├── spinalhdl-idsl-payload_2.12
    │   └── dev
    │       ├── docs
    │       │   ├── spinalhdl-idsl-payload_2.12-javadoc.jar
    │       │   ├── spinalhdl-idsl-payload_2.12-javadoc.jar.md5
    │       │   └── spinalhdl-idsl-payload_2.12-javadoc.jar.sha1
    │       ├── ivys
    │       │   ├── ivy.xml
    │       │   ├── ivy.xml.md5
    │       │   └── ivy.xml.sha1
    │       ├── jars
    │       │   ├── spinalhdl-idsl-payload_2.12.jar
    │       │   ├── spinalhdl-idsl-payload_2.12.jar.md5
    │       │   └── spinalhdl-idsl-payload_2.12.jar.sha1
    │       ├── poms
    │       │   ├── spinalhdl-idsl-payload_2.12.pom
    │       │   ├── spinalhdl-idsl-payload_2.12.pom.md5
    │       │   └── spinalhdl-idsl-payload_2.12.pom.sha1
    │       └── srcs
    │           ├── spinalhdl-idsl-payload_2.12-sources.jar
    │           ├── spinalhdl-idsl-payload_2.12-sources.jar.md5
    │           └── spinalhdl-idsl-payload_2.12-sources.jar.sha1
    ├── spinalhdl-idsl-plugin_2.12
    │   └── dev
    │       ├── docs
    │       │   ├── spinalhdl-idsl-plugin_2.12-javadoc.jar
    │       │   ├── spinalhdl-idsl-plugin_2.12-javadoc.jar.md5
    │       │   └── spinalhdl-idsl-plugin_2.12-javadoc.jar.sha1
    │       ├── ivys
    │       │   ├── ivy.xml
    │       │   ├── ivy.xml.md5
    │       │   └── ivy.xml.sha1
    │       ├── jars
    │       │   ├── spinalhdl-idsl-plugin_2.12.jar
    │       │   ├── spinalhdl-idsl-plugin_2.12.jar.md5
    │       │   └── spinalhdl-idsl-plugin_2.12.jar.sha1
    │       ├── poms
    │       │   ├── spinalhdl-idsl-plugin_2.12.pom
    │       │   ├── spinalhdl-idsl-plugin_2.12.pom.md5
    │       │   └── spinalhdl-idsl-plugin_2.12.pom.sha1
    │       └── srcs
[... 127 lines cut ...]
my enviroment:

> IntelliJ IDEA 2023.3.2 (Community Edition)
> 
> Build #IC-233.13135.103, built on December 20, 2023
> Runtime version: 17.0.9+7-b1087.9 amd64
> VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
> Windows 10.0
> GC: G1 Young Generation, G1 Old Generation
> Memory: 2048M
> Cores: 12
> Registry:
>   ide.experimental.ui=true
>   scala.erase.compiler.process.jdk.once=false
> Non-Bundled Plugins:
>   com.verilang (0.1.8)
>   monokai-pro (1.10)
>   IdeaVIM (2.10.1)
>   PythonCore (233.13135.103)
>   org.exbin.deltahex.intellij (0.2.9)
>   com.alibabacloud.intellij.cosy (1.4.13)
>   org.intellij.scala (2023.3.19)
> Kotlin: 233.13135.103-IJ

[demo1.zip](https://github.com/user-attachments/files/18081265/demo1.zip)

Dolu1990, 2024-12-10:
When you publishLocal the dev, you need before to specify which version of scala.
You project run on scala 2.13
while dev is by default on 2.12

You can do :
sbt clean '++ 2.13.12' publishLocal

Then it should be good.

Readon, 2024-12-10:
> C:\Users\Starx.ivy2\local\com.github.spinalhdl\spinal-core_2.12\dev\ivys\ivy.xml
                        ~~~ I guess a '\' is missing here.

Dolu1990, 2024-12-10:
Hooo, gg ^^ (Starx.ivy2)

facebreeze, 2024-12-11:
Thank you. By specify sbt with scala version solved the problem. 

Looks that `Starx.ivy` is a typo caused by the post, the path in sbt error message is ```C:\Users\Starx\.ivy2\local\com.github.spinalhdl\spinal-core_2.12\dev\ivys\ivy.xml```, and after paste it directly in the post, the `\` got lost.

closed 2024-12-11 by facebreeze