SpinalHDL/SpinalHDL #1613: 『Help』Optimizing Top-Level Connection Logic in SoC Integration(emitFullComponentBindings does not take effect)

dreamflyings, 2024-12-08:
Hello~ @Dolu1990 

I hope that in the top-level module code, there are only signal declarations and connection relationships in module instantiations, without any assignment statements like `always` and `assign`. How can this be achieved? This requirement is useful in specific chip design scenarios, and it also results in high-quality code. It's like the connection of `inout` signals in my demo code, which looks great! I'd like to know how to achieve this for other signals as well. (Setting `emitFullComponentBindings = false` doesn't work, and there are still bugs for complex data structures.)

```scala
package demo.soc

import spinal.core._

object TopLogicOptTest extends App{
 SpinalConfig(targetDirectory = "./verilog/", emitFullComponentBindings = false).generateVerilog(new Component{
 val ta = in (Vec(Vec(Bool(), 2) , 2))
 val tb = out(Vec(Vec(Bool(), 2) , 2))
 val tc = inout(Analog(Vec(Vec(Vec(UInt(2 bits), 2) , 2), 2) ))
 val td = inout(Analog(Vec(Vec(Vec(UInt(2 bits), 2) , 2), 2) ))
 val te = in (Vec(Vec(UInt(2 bits) , 2) , 2))
 val tf = out(Vec(Vec(UInt(2 bits) , 2) , 2))
 val tg = in (Vec(Vec(Vec(UInt(2 bits) , 2) , 2), 2) )
 val th = out(Vec(Vec(Vec(UInt(2 bits) , 2) , 2), 2) )
 val subM = Seq.tabulate(2){i => new Module {
 val io = new Bundle{
 val toTop = new Bundle{
 val ta = in (UInt(2 bits))
 val tb = out(UInt(2 bits))
 val tc = inout(Analog(UInt(8 bits)))
 val td = inout(Analog(UInt(8 bits)))
 val te = in (UInt(4 bits))
 val tf = out(UInt(4 bits))
 val tg = in (UInt(8 bits))
 val th = out(UInt(8 bits))
 }
 val subif = new Bundle{
 val sa = in(Bool())
 val sb = out(Bool())
 val sc = in (UInt(2 bits))
 val sd = out(UInt(2 bits))
 val se = in(Vec(Vec(Vec(UInt(2 bits), 2) , 2), 2) )
 val sf = out(Vec(Vec(Vec(UInt(2 bits), 2) , 2), 2) )
 }
 if(i == 1) subif.flip()
 }
 }.stub().setDefinitionName(s"subTop$i")}
 subM(0).io.subif <> subM(1).io.subif
 Vec(subM.map(_.io.toTop.ta)).assignFromBits(ta.asBits)
 tb.assignFromBits(Vec(subM.map(_.io.toTop.tb)).asBits)
// tc.assignFromBits(Vec(subM.map(_.io.toTop.tc)).asBits)
// td.assignFromBits(Vec(subM.map(_.io.toTop.td)).asBits)
 for(i <- 0 to 1;
 j <- 0 to 1;
 k <- 0 to 1){
 tc(i)(j)(k).setName(s"tc${i}_${2*j+k}")
 td(i)(j)(k).setName(s"td${i}_${2*j+k}")
 tc(i)(j)(k) := subM(i).io.toTop.tc(2*(j*2+k), 2 bits)
 td(i)(j)(k) := subM(i).io.toTop.td(2*(j*2+k), 2 bits)
// subM(i).io.toTop.tc(2*(j*2+k) , 2 bits) := tc(i)(j)(k)
// subM(i).io.toTop.td(2*(j*2+k), 2 bits) := td(i)(j)(k)
 }
 Vec(subM.map(_.io.toTop.te)).assignFromBits(te.asBits)
 tf.assignFromBits(Vec(subM.map(_.io.toTop.tf)).asBits)
 Vec(subM.map(_.io.toTop.tg)).assignFromBits(tg.asBits)
 th.assignFromBits(Vec(subM.map(_.io.toTop.th)).asBits)
 }.setDefinitionName("TopLogicOpt"))
}
```

Dolu1990, 2024-12-09:
Hi,

> How can this be achieved?

What exactly ? 
I mean, do you want : 
- To have an assert if the user specify any logic in the toplevel ?
- Workaround some logic being generated by SpinalHDL in your toplevel, while it shouldn't have done it ?

I'm not sure which one :)

dreamflyings, 2024-12-09:
> Hi,
> 
> > How can this be achieved?
> 
> What exactly ? I mean, do you want :
> 
> * To have an assert if the user specify any logic in the toplevel ?
> * Workaround some logic being generated by SpinalHDL in your toplevel, while it shouldn't have done it ?
> 
> I'm not sure which one :)

Hi~

Long time no see, and thank you so much for your reply!

My request is more aligned with the second point.

I hope the Verilog code generated by SpinalHDL can be more concise. If there is only a connection relationship but no logical operations between the top-level module and the sub-top-level module, the connection relationship should not be generated as separate assign or always statements. Instead, it should be directly placed in the module instantiation like ".a_signal(b_signal)" or ".a_signal({b_signal0, b_signal1, b_signal2})".

Initially, I thought that SpinalHDL couldn't achieve this technically at the moment. However, when I used the "inout" port signal direction type, I found that the generated Verilog instantiation was exactly the connection logic I needed. So I wanted to share this with everyone to see if it's possible to achieve the functionality I need. In my demo code, the lines "val tc = inout(Analog(Vec(Vec(Vec(UInt(2 bits), 2) , 2), 2) )) 
val td = inout(Analog(Vec(Vec(Vec(UInt(2 bits), 2) , 2), 2) ))" were deliberately designed to illustrate my requirements through comparison.

Thank you again for your reply~

dreamflyings, 2024-12-11:
> Ahh i see, Taking the following generated code :
> 
> ```verilog
>   assign _zz_th_0_0_0_1 = _zz_th_0_0_0[7 : 0];
>   assign _zz_th_0_0_0_2 = _zz_th_0_0_0_1[3 : 0];
>   assign _zz_th_0_1_0 = _zz_th_0_0_0_1[7 : 4];
>   assign th_0_1_0 = _zz_th_0_1_0[1 : 0];
>   assign th_0_1_1 = _zz_th_0_1_0[3 : 2];
> ```
> 
> What we can see is _zz_th_0_0_0_1 and _zz_th_0_1_0 are intermediate signals which are used by multiple destinations. I would say, it would not be hard to add an optimization passe to "synthetise" consecttive-static- bit-extractions into a single one (from source to destination).

Wow, that's amazing! Thank you so much!

Seeing such a response from you is truly exhilarating and inspiring!

I suggest adding an optimization option in `SpinalConfig`, keeping it disabled by default. This way, compatibility can be maintained before sufficient validation is done (it can be removed later once everything is complete).

To be frank, I lack the knowledge to fully grasp the concept of the "optimization pass" you mentioned. However, based on my limited understanding, such optimization needs to ensure that the generated Verilog syntax is fully consistent with the internal syntax of SpinalHDL to ensure stability.

Thank you once again! 🙏