SpinalHDL/SpinalHDL #1621: Feature Request: Multi-dimensional `fill` support for `Vec`

szNightFury, 2024-12-13:
Dear @Dolu1990,

I'm writing to respectfully request a new feature for the `Vec` type in SpinalHDL that would significantly improve its usability when working with multi-dimensional structures.

Currently, `Seq` provides a very convenient `fill` method that supports creating multi-dimensional sequences up to 5 dimensions, as shown below:

```scala
  def fill[A](n1: Int, n2: Int)(elem: => A): CC[CC[A]] =
    tabulate(n1)(_ => fill(n2)(elem))

  def fill[A](n1: Int, n2: Int, n3: Int)(elem: => A): CC[CC[CC[A]]] =
    tabulate(n1)(_ => fill(n2, n3)(elem))

  def fill[A](n1: Int, n2: Int, n3: Int, n4: Int)(elem: => A): CC[CC[CC[CC[A]]]] =
    tabulate(n1)(_ => fill(n2, n3, n4)(elem))

  def fill[A](n1: Int, n2: Int, n3: Int, n4: Int, n5: Int)(elem: => A): CC[CC[CC[CC[CC[A]]]]] =
    tabulate(n1)(_ => fill(n2, n3, n4, n5)(elem))
```

However, `Vec` currently only supports a single-dimensional `fill`. To create a multi-dimensional `Vec`, we need to resort to using `Seq.fill` followed by multiple `.map(Vec(_))` calls, which can become quite cumbersome and less readable.

For example, to create a 3x4x5 `Vec` of `UInt(8 bits)`, we currently have to do this:

```scala
val myVec = Seq.fill(3, 4, 5)(UInt(8 bits))).map(d1 => Vec(d1.map(d2 => Vec(d2.map(Vec(_)))))
```
Or
```scala
val myVec = Vec.fill(3)(Vec.fill(4)(Vec.fill(5)(UInt(8 bits))))
```

It would be incredibly helpful if `Vec` could also support a multi-dimensional `fill` similar to `Seq`, allowing us to write:

```scala
val myVec = Vec.fill(3, 4, 5)(UInt(8 bits))
```

This would greatly enhance the clarity and maintainability of code that uses multi-dimensional `Vec`s. I believe this feature would be a valuable addition to SpinalHDL, making it even more powerful and user-friendly.

Thank you for considering this request. I truly appreciate your work on SpinalHDL and your dedication to the community.

Dolu1990, 2024-12-13:
Done :)

closed 2024-12-13 by Dolu1990

szNightFury, 2024-12-30:
Hi!

Following the recent addition of the `Vec.fill` method (which is great!), I was wondering if it would be possible to also add a `tabulate` method to `Vec`, similar to how it works for Scala's `Seq`?

```scala
def tabulate[A](n: Int)(f: Int => A): CC[A] = {
    val b = newBuilder[A]
    b.sizeHint(n)
    var i = 0
    while (i < n) {
      b += f(i)
      i += 1
    }
    b.result()
  }

  /** Produces a two-dimensional $coll containing values of a given function over ranges of integer values starting from 0.
   *  @param   n1  the number of elements in the 1st dimension
   *  @param   n2  the number of elements in the 2nd dimension
   *  @param   f   The function computing element values
   *  @return A $coll consisting of elements `f(i1, i2)`
   *          for `0 <= i1 < n1` and `0 <= i2 < n2`.
   */
  def tabulate[A](n1: Int, n2: Int)(f: (Int, Int) => A): CC[CC[A]] =
    tabulate(n1)(i1 => tabulate(n2)(f(i1, _)))

  /** Produces a three-dimensional $coll containing values of a given function over ranges of integer values starting from 0.
   *  @param   n1  the number of elements in the 1st dimension
   *  @param   n2  the number of elements in the 2nd dimension
   *  @param   n3  the number of elements in the 3nd dimension
   *  @param   f   The function computing element values
   *  @return A $coll consisting of elements `f(i1, i2, i3)`
   *          for `0 <= i1 < n1`, `0 <= i2 < n2`, and `0 <= i3 < n3`.
   */
  def tabulate[A](n1: Int, n2: Int, n3: Int)(f: (Int, Int, Int) => A): CC[CC[CC[A]]] =
    tabulate(n1)(i1 => tabulate(n2, n3)(f(i1, _, _)))

  /** Produces a four-dimensional $coll containing values of a given function over ranges of integer values starting from 0.
   *  @param   n1  the number of elements in the 1st dimension
   *  @param   n2  the number of elements in the 2nd dimension
   *  @param   n3  the number of elements in the 3nd dimension
   *  @param   n4  the number of elements in the 4th dimension
   *  @param   f   The function computing element values
   *  @return A $coll consisting of elements `f(i1, i2, i3, i4)`
   *          for `0 <= i1 < n1`, `0 <= i2 < n2`, `0 <= i3 < n3`, and `0 <= i4 < n4`.
   */
  def tabulate[A](n1: Int, n2: Int, n3: Int, n4: Int)(f: (Int, Int, Int, Int) => A): CC[CC[CC[CC[A]]]] =
    tabulate(n1)(i1 => tabulate(n2, n3, n4)(f(i1, _, _, _)))

  /** Produces a five-dimensional $coll containing values of a given function over ranges of integer values starting from 0.
   *  @param   n1  the number of elements in the 1st dimension
   *  @param   n2  the number of elements in the 2nd dimension
   *  @param   n3  the number of elements in the 3nd dimension
   *  @param   n4  the number of elements in the 4th dimension
   *  @param   n5  the number of elements in the 5th dimension
   *  @param   f   The function computing element values
   *  @return A $coll consisting of elements `f(i1, i2, i3, i4, i5)`
   *          for `0 <= i1 < n1`, `0 <= i2 < n2`, `0 <= i3 < n3`, `0 <= i4 < n4`, and `0 <= i5 < n5`.
   */
  def tabulate[A](n1: Int, n2: Int, n3: Int, n4: Int, n5: Int)(f: (Int, Int, Int, Int, Int) => A): CC[CC[CC[CC[CC[A]]]]] =
    tabulate(n1)(i1 => tabulate(n2, n3, n4, n5)(f(i1, _, _, _, _)))
    ```

Dolu1990, 2025-01-02:
Hi,

Sure it can be added.

szNightFury, 2025-01-03:
> Hi,
> 
> Sure it can be added.

https://github.com/SpinalHDL/SpinalHDL/pull/1643/