SpinalHDL/SpinalHDL #1623: improve signal width inferrence

facebreeze, 2024-12-13:
Hi, 
     for below demo, spinal throw width mismatch error.
```scala
case class Demo() extends Component {
  val io = new Bundle {
    val din = in UInt (4 bits)
    val dout = out UInt (6 bits)
  }
  val factor = 4
  io.dout := io.din * factor
}
```
```WIDTH MISMATCH (6 bits <- 7 bits) on (toplevel/io_dout : out UInt[6 bits]) := (UInt * UInt)[7 bits] at ```

The max value of din is 15, multiply by 4 equals 60, it can fits into 6-bit signal. 

Looks like spinal gets the RHS width by io.din.getWidth + U(4).getWidth = 4 + 3 = 7.

In case of UInt*UInt operation, it's the correct answer, however, for UInt*Int operation, the width should be inferred by `log2Up(io.din.maxValue * factor)`

The problem during project develop is that RHS has to be **resized**, which could potentially obscures existing port width mismatch issue.

Dolu1990, 2024-12-13:
Hi ^^

The issue with changing this behaviour to be smarted, is that it can then have suprising efects with hard to predict widths (from a user perspective).
I mean, this sound scarry to me / deep rabbit hole.
Maybe we could add a UInt * Int operator which explictly behave as you propose ?

On the other hand, there is the AFix type, which kinda do what you mean, it keep track of the values exact range, and use that information to generate operation result range.

facebreeze, 2024-12-13:
It's better to add an 'UInt * Int ' operator.😁

facebreeze, 2024-12-17:
you mean `*^` will be used by a xor based multiplication in the future ? I can't find any usage in current code.

Dolu1990, 2024-12-19:
> you mean *^ will be used by a xor based multiplication in the future ? 

I'm not saying it will be added in the future, but some poeple may understand *^ as a xor based multiplication.

Readon, 2024-12-19:
As we already have below operator. `*^` could be misunderstand compare to `+^`.
I think this function is similar to `#*`, so how about `$*` or `>*` ? I like `>*` which have a direction. ^^

[image]
[image]

facebreeze, 2024-12-19:
an xor based multiplication should use `* ^` instead of `*^`， there is no space between \* and ^ for UInt\*Int operator , so I think it's fine to use \*^ as an operator of UInt\*Int.

Readon, 2024-12-20:
It's not a good practice in engineering to distinguish operation, variable by space.

More important, it is necessary to keep consistent philosophy while designing tools.
There is already an operator works with constant factor `#*`, then we should keep the consistency with the last one of the two chars is `*`.
Selecting just the ahead char is enough, that's why I suggest `>`.

facebreeze, 2024-12-20:
Sounds resonable, and how about `|*` ?    \'|' stands for Int.

Readon, 2024-12-20:
it would be better to avoid chars that already have meanings.'|' is a logic operator represent or.

Readon, 2024-12-20:
how about '*<' which is multiply first , then amplify to following times?

facebreeze, 2024-12-20:
I agree with you.

Readon, 2024-12-20:
> I agree with you.

Nice.^^