SpinalHDL/SpinalHDL #1624: AFix negate sets to minValue when it has a negative number stored

mrberman87, 2024-12-13:
Some pseudo code:

val num = AF(-10, QFormat(16, 8, true))
val invNum = AFix(QFormat(16, 8, true))

invNum := num.negate().fixTo(QFormat(16, 8, true))

invNum will have the value -128

This is working as you would think with positive numbers.  I will try to take a look at this and see if I can't find a solution.

Also, I don't understand why the negate adds a bit.

mrberman87, 2024-12-13:
It appears this was fixed in Dev with (https://github.com/SpinalHDL/SpinalHDL/pull/1547).

@Dolu1990 Can we get a release sometime soon?

Thanks!!

Dolu1990, 2024-12-16:
Hi,

Sure.
Let's sync the release with this merge : 
https://github.com/SpinalHDL/SpinalHDL/pull/1625

mrberman87, 2024-12-16:
Sounds good, Thanks!!!