SpinalHDL/SpinalHDL #1634: Byte Mask Write for URAM Synthesized as Multiple BRAMs Instead of Single URAM

szNightFury, 2024-12-27:
Hi @Dolu1990, I am testing the byte mask write functionality for URAM (UltraRAM) in SpinalHDL. I expect the generated Verilog to infer a single URAM with byte write enables. However, the current implementation synthesizes the memory as nine separate BRAMs (Block RAMs) when using a byte mask for writing.

**Steps to Reproduce:**

1. Use the following SpinalHDL code to define a memory with byte mask write:

    ```scala
    class ByteMaskWrite() extends Component {
      val io = new Bundle {
        val wCmd = slave(Flow(TupleBundle(UInt(12 bits), UInt(72 bits), Bits(9 bits))))
        val rCmd = slave(Stream(UInt(12 bits)))
        val rRsp = master(Stream(UInt(72 bits)))
      }

      val mem = Mem(UInt(72 bits), 4096)

      mem.write(
        address = io.wCmd._1,
        data = io.wCmd._2,
        enable = io.wCmd.valid,
        mask = io.wCmd._3
      )

      io.rCmd.ready := io.rRsp.isFree
      io.rRsp.valid := RegNextWhen(io.rCmd.valid, io.rCmd.ready, False)
      io.rRsp.payload := mem.readSync(
        address = io.rCmd.payload,
        enable = io.rCmd.ready
      )
    }
    ```

2. Generate Verilog using SpinalHDL.

**Expected Behavior:**

The generated Verilog should infer a single URAM with appropriate byte write enables, utilizing the `mem_symbol` array as shown in the modified Verilog code below.

**Actual Behavior:**

The generated Verilog creates nine separate `reg [7:0]` arrays (`mem_symbol0` to `mem_symbol8`), which leads the synthesis tool to infer nine separate BRAMs instead of a single URAM.

**Generated Verilog (Incorrect):**

```verilog
// Generator : SpinalHDL v1.10.2a    git head : a348a60b7e8b6a455c72e1536ec3d74a2ea16935
// Component : ByteMaskWrite
// Git hash  : b5c6d8a2e6eccf2a0c5f53c1a9e37d8c967fd903

`timescale 1ns/1ps

module ByteMaskWrite (
  input  wire          io_wCmd_valid,
  input  wire [11:0]   io_wCmd_payload_1,
  input  wire [71:0]   io_wCmd_payload_2,
  input  wire [8:0]    io_wCmd_payload_3,
  input  wire          io_rCmd_valid,
  output wire          io_rCmd_ready,
  input  wire [11:0]   io_rCmd_payload,
  output wire          io_rRsp_valid,
  input  wire          io_rRsp_ready,
  output wire [71:0]   io_rRsp_payload,
  input  wire          clk,
  input  wire          reset
);

  reg        [71:0]   mem_spinal_port1;
  wire       [71:0]   _zz_mem_port;
  wire                io_rRsp_isFree;
  reg                 io_rCmd_valid_regNextWhen;
  reg [7:0] mem_symbol0 [0:4095];
  reg [7:0] mem_symbol1 [0:4095];
  reg [7:0] mem_symbol2 [0:4095];
  reg [7:0] mem_symbol3 [0:4095];
  reg [7:0] mem_symbol4 [0:4095];
  reg [7:0] mem_symbol5 [0:4095];
  reg [7:0] mem_symbol6 [0:4095];
  reg [7:0] mem_symbol7 [0:4095];
  reg [7:0] mem_symbol8 [0:4095];
  reg [7:0] _zz_memsymbol_read;
  reg [7:0] _zz_memsymbol_read_1;
  reg [7:0] _zz_memsymbol_read_2;
  reg [7:0] _zz_memsymbol_read_3;
  reg [7:0] _zz_memsymbol_read_4;
  reg [7:0] _zz_memsymbol_read_5;
  reg [7:0] _zz_memsymbol_read_6;
  reg [7:0] _zz_memsymbol_read_7;
  reg [7:0] _zz_memsymbol_read_8;

  assign _zz_mem_port = io_wCmd_payload_2;
  always @(*) begin
    mem_spinal_port1 = {_zz_memsymbol_read_8, _zz_memsymbol_read_7, _zz_memsymbol_read_6, _zz_memsymbol_read_5, _zz_memsymbol_read_4, _zz_memsymbol_read_3, _zz_memsymbol_read_2, _zz_memsymbol_read_1, _zz_memsymbol_read};
  end
  always @(posedge clk) begin
    if(io_wCmd_payload_3[0] && io_wCmd_valid) begin
      mem_symbol0[io_wCmd_payload_1] <= _zz_mem_port[7 : 0];
    end
    if(io_wCmd_payload_3[1] && io_wCmd_valid) begin
      mem_symbol1[io_wCmd_payload_1] <= _zz_mem_port[15 : 8];
    end
    if(io_wCmd_payload_3[2] && io_wCmd_valid) begin
      mem_symbol2[io_wCmd_payload_1] <= _zz_mem_port[23 : 16];
    end
    if(io_wCmd_payload_3[3] && io_wCmd_valid) begin
      mem_symbol3[io_wCmd_payload_1] <= _zz_mem_port[31 : 24];
    end
    if(io_wCmd_payload_3[4] && io_wCmd_valid) begin
      mem_symbol4[io_wCmd_payload_1] <= _zz_mem_port[39 : 32];
    end
    if(io_wCmd_payload_3[5] && io_wCmd_valid) begin
      mem_symbol5[io_wCmd_payload_1] <= _zz_mem_port[47 : 40];
    end
    if(io_wCmd_payload_3[6] && io_wCmd_valid) begin
      mem_symbol6[io_wCmd_payload_1] <= _zz_mem_port[55 : 48];
    end
    if(io_wCmd_payload_3[7] && io_wCmd_valid) begin
      mem_symbol7[io_wCmd_payload_1] <= _zz_mem_port[63 : 56];
    end
    if(io_wCmd_payload_3[8] && io_wCmd_valid) begin
      mem_symbol8[io_wCmd_payload_1] <= _zz_mem_port[71 : 64];
    end
  end

  always @(posedge clk) begin
    if(io_rCmd_ready) begin
      _zz_memsymbol_read <= mem_symbol0[io_rCmd_payload];
      _zz_memsymbol_read_1 <= mem_symbol1[io_rCmd_payload];
      _zz_memsymbol_read_2 <= mem_symbol2[io_rCmd_payload];
      _zz_memsymbol_read_3 <= mem_symbol3[io_rCmd_payload];
      _zz_memsymbol_read_4 <= mem_symbol4[io_rCmd_payload];
      _zz_memsymbol_read_5 <= mem_symbol5[io_rCmd_payload];
      _zz_memsymbol_read_6 <= mem_symbol6[io_rCmd_payload];
      _zz_memsymbol_read_7 <= mem_symbol7[io_rCmd_payload];
      _zz_memsymbol_read_8 <= mem_symbol8[io_rCmd_payload];
    end
  end

  assign io_rRsp_isFree = ((! io_rRsp_valid) || io_rRsp_ready);
  assign io_rCmd_ready = io_rRsp_isFree;
  assign io_rRsp_valid = io_rCmd_valid_regNextWhen;
  assign io_rRsp_payload = mem_spinal_port1;
  always @(posedge clk or posedge reset) begin
    if(reset) begin
      io_rCmd_valid_regNextWhen <= 1'b0;
    end else begin
      if(io_rCmd_ready) begin
        io_rCmd_valid_regNextWhen <= io_rCmd_valid;
      end
    end
  end

endmodule
```

**Modified Verilog:**

```verilog
// Generator : SpinalHDL v1.10.2a    git head : a348a60b7e8b6a455c72e1536ec3d74a2ea16935
// Component : ByteMaskWrite
// Git hash  : b5c6d8a2e6eccf2a0c5f53c1a9e37d8c967fd903

`timescale 1ns/1ps

module ByteMaskWrite (
  input  wire          io_wCmd_valid,
  input  wire [11:0]   io_wCmd_payload_1,
  input  wire [71:0]   io_wCmd_payload_2,
  input  wire [8:0]    io_wCmd_payload_3,
  input  wire          io_rCmd_valid,
  output wire          io_rCmd_ready,
  input  wire [11:0]   io_rCmd_payload,
  output wire          io_rRsp_valid,
  input  wire          io_rRsp_ready,
  output wire [71:0]   io_rRsp_payload,
  input  wire          clk,
  input  wire          reset
);

  reg        [71:0]   mem_spinal_port1;
  wire                io_rRsp_isFree;
  reg                 io_rCmd_valid_regNextWhen;
  reg [71:0] mem_symbol [0:4095];
  integer i;

  always @(posedge clk) begin
    if (io_wCmd_valid) begin
      for (i = 0; i < 9; i = i + 1) begin
        if (io_wCmd_payload_3[i]) begin
          mem_symbol[io_wCmd_payload_1][i*8 +: 8] <= io_wCmd_payload_2[i*8 +: 8];
        end
      end
    end
  end

  assign io_rRsp_isFree = ((! io_rRsp_valid) || io_rRsp_ready);
  assign io_rCmd_ready = io_rRsp_isFree;
  assign io_rRsp_valid = io_rCmd_valid_regNextWhen;
  assign io_rRsp_payload = mem_spinal_port1;

  always @(posedge clk) begin
      if(io_rCmd_ready) begin
        mem_spinal_port1 <= mem_symbol[io_rCmd_payload];
      end
    end

  always @(posedge clk or posedge reset) begin
    if(reset) begin
      io_rCmd_valid_regNextWhen <= 1'b0;
    end else begin
      if(io_rCmd_ready) begin
        io_rCmd_valid_regNextWhen <= io_rCmd_valid;
      end
    end
  end

endmodule
```

**Analysis:**

It seems the current implementation of byte mask writes in SpinalHDL might be leading to this behavior. The way separate reg arrays are generated for each byte slice appears to encourage the synthesis tool to infer multiple BRAMs instead of a single URAM.

**Possible Approach:**

One potential way to address this could be to explore alternative code generation strategies for memory writes with masks. Perhaps avoiding bit slicing and grouping reg for each byte, similar to what's demonstrated in the modified Verilog, might help. In the modified code, a single reg [71:0] mem_symbol [0:4095] array represents the memory, and a for loop with bit slicing within the always @(posedge clk) block handles byte-wise writing based on the mask. This seems to guide the synthesis tool towards inferring a URAM correctly.

**Environment:**

SpinalHDL version: v1.10.2a
Synthesis tool: Vivado 2024.2

Thank you for considering this request. I truly appreciate your work on SpinalHDL and your dedication to the community.

Dolu1990, 2024-12-28:
Hi,
synthesis tools infer RW ports with byte mask using various incompatible Verilog patterns. There is no universal way to handle it. So SpinalHDL fallback to spliting each byte lane into its own ram to ensure it get inferred (in some non optimal ways)

I would say, when it comes to specific target, the best maybe to blackbox the ram.

myRam.generateAsBlackBox()

szNightFury, 2024-12-30:
> Hi, synthesis tools infer RW ports with byte mask using various incompatible Verilog patterns. There is no universal way to handle it. So SpinalHDL fallback to spliting each byte lane into its own [...]

Hi,

Thanks for the quick reply and the explanation! I get the whole mess with different tools and their Verilog quirks for byte-masked RAMs. Splitting the lanes makes sense to guarantee it works, even if it's not always pretty.

I tried out `myRam.generateAsBlackBox()` like you said, and it's definitely helping with the URAM inference, so that's great!  But I'm a bit stuck on the blackbox mapping.

So, it looks like `generateAsBlackBox()` just auto-maps based on the read/write port count to some built-in blackboxes like `Ram_1r_1ws`, right? I'm hitting some snags where parameters (like 'technology') that aren't defined in my `Ram_1r_1ws.v` (and honestly, I don't care about them), and it's throwing errors.

My question is: is there an easy way to figure out exactly what parameters the mapped blackbox needs, without digging through the library's blackbox code? Or, if there are mismatches like this 'technology' thing, is my best bet to just write a completely custom blackbox that perfectly matches my RAM's needs?

Any tips would be super helpful!  Thanks again for all your work on SpinalHDL!

Dolu1990, 2025-01-02:
> I'm hitting some snags where parameters (like 'technology') that aren't defined in my Ram_1r_1ws.v (and honestly, I don't care about them), and it's throwing errors

Just ignore the technology parameter, it isn't changing the behaviour, it is more a potential helper to make things mapping better. In practice it is never used.
What error do you get ?

Here is an example of verilog blackboxes : 
https://github.com/litex-hub/pythondata-cpu-vexiiriscv/tree/main/pythondata_cpu_vexiiriscv/verilog

(They are kinda tunned to work with byte mask, and will not work for all cases)

szNightFury, 2025-01-03:
> Just ignore the technology parameter, it isn't changing the behaviour, it is more a potential helper to make things mapping better. In practice it is never used. What error do you get ?
>
> Here is an [...]

Hi,

Thank you for your detailed response and for pointing me toward the example Verilog blackboxes and additional insights. Now, I’m facing anthoer challenge. I’m trying to use `mem.generateAsBlackBox()` in combination with `readSync` and `m2sPipe` to automatically generate a blackbox with a two-stage pipeline. I came across some mentions of this in [[issue #1392](https://github.com/SpinalHDL/SpinalHDL/issues/1392)](https://github.com/SpinalHDL/SpinalHDL/issues/1392), but I’m a bit unclear on the exact implementation steps since it seems that related parameters like `rdLatency` and `membufferphase` have been removed.

Could you provide some guidance or examples on how to ensure that the output registers are integrated into the BRAM blackbox while using m2sPipe effectively?

Thanks again for all your help and the incredible work on SpinalHDL!

Dolu1990, 2025-01-03:
So, https://github.com/SpinalHDL/SpinalHDL/issues/1392 isn't merged in dev. 
Which mean you need to get the right SpinalHDL branch, and then also add  config.addTransformationPhase(new MemReadBufferPhase) to your SpinalConfig to enable the feature.
Then it should be good.
I will open a PR for that feature.

Dolu1990, 2025-01-03:
https://github.com/SpinalHDL/SpinalHDL/pull/1642

szNightFury, 2025-01-03:
> So, #1392 isn't merged in dev. Which mean you need to get the right SpinalHDL branch, and then also add config.addTransformationPhase(new MemReadBufferPhase) to your SpinalConfig to enable the feature. Then it should be good. I will open a PR for that feature.

Nice :) and I was wondering—since the Ram_1w_1rs blackbox now supports rdLatency, wouldn’t it make sense to adapt the same for other blackboxes with read ports as well? It seems like a useful and consistent feature to have across the board.

Dolu1990, 2025-01-03:
> Would it be worth considering extending this feature to other blackboxes with read ports as well? 

Yes

I merged the PR.