SpinalHDL/SpinalHDL #1635: how to close the "No drive on" error detect for component input

littlezpf666, 2024-12-31:
Does SpinalHDL have some attributes or function to close the "No drive on" error detect for specific input of Blackbox/ component? 

For example, in following case, what should I do, if I want to dangle the input of Blackbox?
class Ram_1w_1r(wordWidth: Int, wordCount: Int) extends BlackBox {
  // Add VHDL Generics / Verilog parameters to the blackbox
  // You can use String, Int, Double, Boolean, and all SpinalHDL base
  // types as generic values
  addGeneric("wordCount", wordCount)
  addGeneric("wordWidth", wordWidth)

  // Define IO of the VHDL entity / Verilog module
  val io = new Bundle {
    val clk = in Bool()
    val wr = new Bundle {
      val en   = in Bool()
      val addr = in UInt (log2Up(wordCount) bits)
      val data = in Bits (wordWidth bits)
    }
    val rd = new Bundle {
      val en   = in Bool()
      val addr = in UInt (log2Up(wordCount) bits)
      val data = out Bits (wordWidth bits)
    }
  }

Readon, 2024-12-31:
Could assign some default value while using this blackbox helps?

littlezpf666, 2025-01-01:
I want to develop a SOC building tool by SpinalHDL and scala. In some of my scenario, the Blackbox input don't know the input connection information. assigning the default value is not a good way for me. So, I want to know whether the "No drive on" error detect for component input can be closed according to demand.

Dolu1990, 2025-01-01:
Ahhh you want to keep an input of the blackbox floating ? And SpinalHDL to ignore it (no error generated) ?

littlezpf666, 2025-01-01:
Yes, the Blackbox input status is like the visualized IP in vivado. Although the IP is generated, the input/output of component is floating.
[image]

Dolu1990, 2025-01-01:
I pushed the feature in dev just now.

mySignal..addTag(allowFloating)  would do the trick.

closed 2025-01-01 by Dolu1990

Dolu1990, 2025-01-01:
Note it generates : 

```verilog
  wire                bb_xxx;

  unamed bb (
    .xxx (bb_xxx)  //i
  );
``` 
  
  But maybe you wanted 
  
  ```verilog
  unamed bb (
    .xxx ()  //i  
  );
``` 

 ?

littlezpf666, 2025-01-02:
Yes, I want spinalhdl allow to generate this verilog result. 
unamed bb (
  .xxx ()  //i  
);

littlezpf666, 2025-01-02:
so now if I want to float the port of blackbox, I can do like below?

class Ram_1w_1r(wordWidth: Int, wordCount: Int) extends BlackBox {
  // Add VHDL Generics / Verilog parameters to the blackbox
  // You can use String, Int, Double, Boolean, and all SpinalHDL base
  // types as generic values
  addGeneric("wordCount", wordCount)
  addGeneric("wordWidth", wordWidth)

  // Define IO of the VHDL entity / Verilog module
  val io = new Bundle {
    val clk = in Bool()
    val wr = new Bundle {
      val en   = in Bool().addTag(allowFloating)
      val addr = in UInt (log2Up(wordCount) bits).addTag(allowFloating)
      val data = in Bits (wordWidth bits).addTag(allowFloating)
    }
    val rd = new Bundle {
      val en   = in Bool().addTag(allowFloating)
      val addr = in UInt (log2Up(wordCount) bits).addTag(allowFloating)
      val data = out Bits (wordWidth bits).addTag(allowFloating)
    }
  }

Dolu1990, 2025-01-02:
Fixed ^^

Dolu1990, 2025-01-02:
> so now if I want to float the port of blackbox, I can do like below?

Yes

commit c6dcae63 (2025-01-01): fix #1635 add allowFloating
commit 3819ba36 (2025-01-01): fix #1635 add allowFloating
commit 82c5c6d1 (2025-01-02): Merge pull request #1639 from SpinalHDL/1635

fix #1635
commit 9b999d59 (2025-01-02): fix #1635