SpinalHDL/SpinalHDL #1640: [Axi4Master] Here's a bug in Axi4Master.writeSingle

m1506789861, 2025-01-02:
val strb = if (len == 0) {
            ((BigInt(1) << realLen) - 1) << padFront
          } else beat match {
            case 0 => (BigInt(1) << (bytePerBeat - padFront)) - 1
            case `len` => ~((BigInt(1) << padBack) - 1)
            case _ => BigInt(1) << busConfig.bytePerWord
          }

should be: 

            val strb = if (len == 0) {
                ((BigInt(1) << realLen) - 1) << padFront
            } else beat match {
                case 0 => (BigInt(1) << (bytePerBeat - padFront)) - 1
                case `len` => (BigInt(1) << (busConfig.bytePerWord - padBack)) - 1
                case _ => (BigInt(1) << busConfig.bytePerWord) - 1
            }

Dolu1990, 2025-01-06:
Hi,

Thanks !
@KireinaHoro :)

KireinaHoro, 2025-01-06:
Cool that you are using this!  Can you detail a bit more (and maybe provide a waveform where the master generated incorrect signals)?

I get that the `case _` is indeed wrong since with the -1 missing, it's sending all zeroes for `strb`.  The ``case `len` `` is more complicated:

- when `busConfig.bytePerWord == bytePerBeat` (which is `1 << size`): we are sending full beats.  `~((1 << padBack) - 1)` should be the same as `(1 << (bytePerBeat - padBack)) - 1` after truncation
- when they are different: this is a narrow burst.  We should always use `bytePerBeat` since the beat contains less bytes than `bytePerWord`; but yes your way of writing it is better since in my case this will result in more bytes being transmitted than necessary.

This does show that the simulation master requires more test cases to make it really bullet proof...

m1506789861, 2025-01-06:
When `len` >= 2, scala will report compilation error. 

For example, if len = 2,
When beat == 1, "BigInt(1) << busConfig.bytePerWord " will be assigned to strb, so strb will too large, and is also wrong.
When beat == 2, "~((BigInt(1) << padBack) - 1) " will be assigned to strb, so strb's high bits will be all 1, also too large because it is BigInt.

These are the problems.

KireinaHoro, 2025-01-06:
I don't understand -- what do you mean with "Scala compilation error"?  Can you post the error?  Would be great if you can also post a small example so I can play with it a bit.

My understanding was that assigning with `#=` will automatically truncate if the `BigInt` does not fit inside the signal (by SpinalSim), but I might be wrong here.

m1506789861, 2025-01-06:
**For example:**

            val axi4Master = Axi4Master(dut.io.input, dut.clockDomain)
            axi4Master.reset()

            val axi4Memory = AxiMemorySim(
                axi         = dut.io.output,
                clockDomain = dut.clockDomain,
                config      = AxiMemorySimConfig(
                    maxOutstandingReads  = 8,
                    maxOutstandingWrites = 8,
                    readResponseDelay    = 0,
                    writeResponseDelay   = 0,
                    interruptProbability = 0,
                    interruptMaxDelay    = 0,
                    defaultBurstType     = 1,
                    useAlteraBehavior    = false
                )
            )
            axi4Memory.reset()
            dut.clockDomain.waitRisingEdge()

            //write
            val wrBaseAddr = 0xFF00
            val wrData     = (0 until 17).map(_.toByte).toList
            val wrBurstLen = ceil(wrData.length.toDouble / dut.axi4Config.bytePerWord).toInt - 1
            axi4Master.write(address = wrBaseAddr, data = wrData, burst = Axi4Bursts.Incr, len = wrBurstLen, size = log2Up(dut.axi4Config.bytePerWord))

**will report:**

[Progress] Start TestDemo test simulation with seed 1811711499
Handling AXI4 Master read cmds...
Handling AXI4 Master read resp...
Handling AXI4 Master write cmds...
Handling AXI4 Master write...
Axi4Master (unnamed) [AW]	: addr 0xff00 size 3 len 2 burst Incr
Axi4Master (unnamed) [W]	: data 0001020304050607 strb 0xff last false
Page fault while reading page 0 (0x0)
Adding page 0 at 0x0
Axi4Master (unnamed) [W]	: data 08090a0b0c0d0e0f strb 0x100 last false

ASSIGNMENT ERROR : -128 is outside the range of TestDemo/io_input_w_payload_strb : Bits[8 bits]
Exception in thread "main" java.lang.Exception
	at spinal.sim.SimError$.apply(Misc.scala:12)
	at spinal.sim.DataType.rangeError(Signal.scala:13)
	at spinal.sim.BitsDataType.checkBigIntRange(Signal.scala:75)
	at spinal.sim.SimManager.setBigInt(SimManager.scala:182)
	at spinal.core.sim.package$.setBigInt(package.scala:166)
	at spinal.core.sim.package$SimBitVectorPimper.$hash$eq(package.scala:422)
	at spinal.lib.bus.amba4.axi.sim.Axi4Master$$anonfun$writeSingle$5$$anonfun$apply$6$$anonfun$apply$7.apply(Axi4Master.scala:301)
	at spinal.lib.bus.amba4.axi.sim.Axi4Master$$anonfun$writeSingle$5$$anonfun$apply$6$$anonfun$apply$7.apply(Axi4Master.scala:291)
	at spinal.lib.bus.amba4.axi.sim.Axi4Master$$anonfun$7.apply(Axi4Master.scala:324)
	at spinal.lib.bus.amba4.axi.sim.Axi4Master$$anonfun$7.apply(Axi4Master.scala:322)
	at spinal.lib.sim.StreamDriver.fsm(Stream.scala:122)

KireinaHoro, 2025-01-06:
Yeah ok it looks I'm wrong -- `#=` does check the number of bits.

Can you prepare a PR?  But please don't use `busConfig.bytePerWord` -- use `bytePerBeat` to accomodate for narrow bursts.  Thanks for the catch!

m1506789861, 2025-01-06:
If it's convenient, could you please fix the code using "bytePerBeat". Thanks!

hy0000, 2025-11-25:
> 
> The interface is designed to allow this (note the `maxLen` argument), but I don’t have such a use case locally. I’d be happy to review a PR!

Have you ever tried reg access using Axi4Master,  like writing 4B on 128b AXI:
```scala
val data = List(1, 2, 3, 4).map(_.toByte)
writeSingle(0x0C, data, len = 0, size = 2)
```
The except value on AXI in my view would be:
```
awaddr = 0x0C
awsize = 2
awlen = 0
wdata = 0x4321_0000_0000_0000
wstrb = 0xF000
```
But i got assertion failed. spinal verison is 1.13.0

KireinaHoro, 2025-11-25:
No I have not, as I said I don’t have a use case for narrow writes.  You are very welcome to propose a fix — I’d be happy to review it.