SpinalHDL/SpinalHDL #1644: Simulation Kernel Library Not Found (`librdi_simulator_kernel.dll`)

skylayer, 2025-01-05:
#### Description:
Simulation fails to load the required shared library `librdi_simulator_kernel.dll`. This may be due to [the recent renaming of the kernel library in Vivado 2024.2](https://adaptivesupport.amd.com/s/article/000037038).

#### System Information:
- **OS:** Windows  
- **JDK Version:** OpenJDK 23.0.1  
- **SpinalHDL Git Head:** `8c0ad67d2e84a033dcb7ad95edc1f28526a547b0`  
- **Vivado Version:** 2024.2  

#### Error Message:
```
Could not load simulation kernel library (librdi_simulator_kernel.dll): Failed to load shared library "librdi_simulator_kernel.dll". 指定的模块未找到。
```

#### Exception:
```
Exception in thread "main" java.lang.Exception: ISim engine error: Failed to Load up XSI.
	at spinal.sim.xsi.JNIXSIIfaceJNI.new_XSIIface(Native Method)
	at spinal.sim.xsi.XSIIface.<init>(XSIIface.java:40)
	at spinal.sim.XSimBackend.getInterface(XSimBackend.scala:361)
	at spinal.sim.SimXSim.<init>(SimXSim.scala:9)
	at spinal.core.sim.SpinalSimConfig$$anon$7.newSimRaw(SimBootstraps.scala:1203)
	at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:616)
	at spinal.core.sim.SimCompiled.doSim(SimBootstraps.scala:586)
```

#### Suggested Fix:
Update the library reference in SpinalHDL to support the new library name (`xv_simulator_kernel.dll`) introduced in Vivado 2024.2.

skylayer, 2025-01-05:
I'm unsure whether we can address this by detecting the Vivado version or by verifying the presence of either the old or new dynamic library. However, both methods could be considered inelegant solutions.

Dolu1990, 2025-01-06:
I would say, detecting the vivado version would be the way, as it could be used to workaround most flow variations.

Pyromuffin, 2025-02-11:
having this issue as well. making a local build with the new xv_simulator_kernel.dll path seems to fix the issue, though only if you don't need to use older versions of vivado.