SpinalHDL/SpinalHDL #1647: i found a problem that really vex me

dzjdegithub, 2025-01-09:
if I use "in" or "out" for signal name,that will be conflict with the IO port direction declarition,could you please fix this?thank you so much.

dzjdegithub, 2025-01-09:
by the way,i like spinal so much ,i think its better than chisel

davine47, 2025-01-09:
Have a try on this code segment, using withoutReservedKeywords: 
```
class Test extends Component {
  val io = new Bundle {
    val in = spinal.core.in Bool()
    val out = spinal.core.out Bool()
  }
  noIoPrefix()
  withoutReservedKeywords = true
  io.in.setName("in")
  io.out.setName("out")
  io.out := False | io.in
}
```
```
// Generator : SpinalHDL v1.11.0    git head : 63852c61e498798f4e293594ce53fcb02c45eb6b
// Component : Test
// Git hash  : 66d3f6fe7610ffb20428efbb229992b64174d112

`timescale 1ns/1ps 
module Test (
  input  wire          in,
  output wire          out
);
  assign out = (1'b0 || in);
endmodule```

closed 2025-01-09 by dzjdegithub