SpinalHDL/SpinalHDL #1649: i have a question

dzjdegithub, 2025-01-15:
in spinal..lib.fsm  how can i  get the current state and nextstate ?and how can i get the statecode? for exmaple, if i  got a signal named out,the out shoule be set to the high if the state is bigger than previous state,so i need current state and nextstate and their binary code to compare.And thank you for your patience to answer this question.

hhucchenyixiao, 2025-02-21:
> Hmm it isn't realy made for that kind of usage. I would say, you would need to explicitly call the FSM build function, then you can preceduraly analyse the FSM datamodel to generate the specific hardware you need

Can we make `fsm.State` convertible to Bits? Although it's possible to output the `stateReg` value externally using `post_build` now, it can't be output like the example below:

```
fsm.postBuild{
    io.fsm_is_state_a := fsm.stateReg.asBits === fsm.stateA.asBits
}
```

mrberman87, 2025-02-21:
You could do something like:

```
val io = new Bundle {
  val isSecond = out(Bool())
}

val stateMachine = new StateMachine {
  val first  = new State with EntryPoint
  val second = new State
  ...
}

io.isSecond := stateMachine.isActive(stateMachine.second)
```

hhucchenyixiao, 2025-02-21:
Thank you for your help, but this approach does not cover all scenarios. For example, there is still no suitable way to output `fsm_next_is_state_a` like this:

```
fsm.postBuild{
    io.fsm_next_is_state_a := fsm.stateNext.asBits === fsm.stateA.asBits
}
```