SpinalHDL/SpinalHDL #1664: Multi-Clock Domain Assignments

ChomperT, 2025-02-16:
I have two components, A and B.  Each component has two clocks.  I want to 'relate' (or 'connect') B's clkA and clkB within A. How do I do this?

> 
case class B() extends Component {
...
  val clkA = ClockDomain.external("clkA")
  val clkB = ClockDomain.external("clkB")
...
}
case class A() extends Component {
  val clkC = ClockDomain.external("clkC")
  val clkD = ClockDomain.external("clkD")
  val b = B()
  b.clkA = clkC //?
  b.clkB = clkD //?
}

Dolu1990, 2025-02-16:
Are clkA and clkC different clock domains ? or they are the same ?

ChomperT, 2025-02-16:
> Are clkA and clkC different clock domains ? or they are the same ?

different clock domain

ChomperT, 2025-02-16:
> Are clkA and clkC different clock domains ? or they are the same ?

same

Dolu1990, 2025-02-16:
Then 

```scala
case class B(clkA : ClockDomain, clkB : ClockDomain) extends Component {

}
case class A() extends Component {
val clkC = ClockDomain.external("clkC")
val clkD = ClockDomain.external("clkD")
val b = B(clkA, clkB)
}
```

ChomperT, 2025-02-16:
Thank you so much for your response! I was asking because I have a rather complex Component with fixed parameters, and I was just exploring whether there was any way to connect different clock domains without modifying the Component's parameters. Your reply made it clear to me.

closed 2025-02-17 by Dolu1990