SpinalHDL/SpinalHDL #1670: Can I run simulations with SpinalSim without recompiling the DUT

HC153Yuanyuan, 2025-03-04:
My DUT compilation takes quite a long time, and when debugging the simulation environment, I don't want to spend time waiting for the DUT to recompile.

Dolu1990, 2025-03-04:
Hi,

In the exact case were you only chane the scala testbench code and rerunning it ?

HC153Yuanyuan, 2025-03-04:
Yes, that's the case.

Dolu1990, 2025-03-04:
The only way to get close to this so far would be to define define a toplevel which blackbox the pregenerated design you want to test, and then running your testbench against that toplevel.
Not great but that would work, as SpinalHDL would only need to generate the little dummy toplevel, and the verilator compilation is already cached by SpinalHDL

HC153Yuanyuan, 2025-03-04:
Thank you very much! It’s indeed a way, though not perfect. I’ll give it a try.

closed 2025-03-04 by HC153Yuanyuan