SpinalHDL/SpinalHDL #1672: How to make flowdriver send data at regular intervals?

buloruabutata, 2025-03-06:
It seems to rewrite transactionDelay, but I don't know scala very well.
```
FlowDriver(dut.io.BAD_in(i), dut.clockDomain) {
	override var transactionDelay: () => Int = () => 1
} { payload =>

} 
```

buloruabutata, 2025-03-06:
with the help of AI, I solved the above problem, but Flowdriver still delays for random periods before sending the first data.
```
FlowDriver(dut.io.BAD_in(i), dut.clockDomain) {
	override var transactionDelay: () => Int = () => 1
} { payload =>

} .transactionDelay = () => {
	1
} 
```

Dolu1990, 2025-03-10:
Hmm 
Instead of override var transactionDelay: () => Int = () => 1
setFactor(2.0f) should od the trick

buloruabutata, 2025-03-11:
Thank you. flow_driver.setFactor(2.0f) behaves similar to the following code, and is more concise.

```
val flow_driver = FlowDriver(...）{...}
flow_driver.transactionDelay = () => {
	0 
}
flow_driver.delay = 0
```

closed 2025-03-11 by buloruabutata