SpinalHDL/SpinalHDL #1684: how to use [T <: Data] with dynamic width

buloruabutata, 2025-03-19:
I want to make all ports in the project UInt or SInt and declare the bit width dynamically, but if I use generics, I have to pass in generic parameters for each different bit width.

I want to pass in SInt or UInt and declare the bit width when creating the port, instead of passing in SInt(8 bits)

buloruabutata, 2025-03-19:
Just like #define data_type SInt in C language

Dolu1990, 2025-03-19:
HardType should solve it https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Other%20language%20features/utils.html#the-safe-way

buloruabutata, 2025-03-19:
HardType works like this
```
case class ShiftRegister[T <: Data](dataType: HardType[T], depth: Int) extends Component {
  val io = new Bundle {
    val input  = in (dataType())
    val output = out(dataType())
  }
  // ...
}
val shiftReg = ShiftRegister(Bits(32 bits), depth = 8)
```
I hope it's like below

```
case class ShiftRegister[T <: Data](dataType, depth: Int) extends Component {
  val io = new Bundle {
    val input  = in (dataType(16 bits))
    val output = out(dataType(32 bits))
  }
  // ...
}
val shiftReg = ShiftRegister(Bits, depth = 8)
```

Dolu1990, 2025-03-19:
Ahhh
It may be possible using T <: BitVector  + also having the hardtype, but then setting the bits width using dataType().setWidth(16)

kinda dirty.