SpinalHDL/SpinalHDL #1685: Need help debugging a deadlock in Tilelink Hub

petrmikheev, 2025-03-19:
Problem: GET request comes to `hub_1/io_up_a`, but doesn't appear on `hub_1/io_down_a`.

[image]
[dump.vcd](https://cloud.mikhe.ch/index.php/s/u65RAYJSY8CFVnL)

This image shows two 64 byte read requests coming from VexiiRiscv lsuL1Bus. The first one successfully finishes, the second one (same request to the next 64 byte block) for some reason can't go through the hub.

The hub has two masters (vexiiriscv.lsuL1Bus and vexiiriscv.iBus) and two slaves:
- ROM at 0x40000000 - 0x40000200 (iBus reads only from there and only a few times in the beginning)
- RAM at 0x80000000 - 0xc0000000 (manually written adapter from Tilelink to DDR3 Soft Controller IP block available for my FPGA)

More details:
- I haven't managed to reproduce the problem in simulation. The dump is obtained right from FPGA (Efinix Ti60) using Efinity Debugger.
- The program in ROM first writes 32KB to RAM at 0x80000000 - 0x80008000, then reads the same data sequentially. The deadlock happens on every run. It happens on the reading stage, always at address 0x80004240.

Maybe I can't reproduce it in simulation because the DDR3 controller IP block comes without sources, and my simplified kind-of-simulation-model has different timings.

I am almost sure that the problem is on my side, but I have no idea where to look. In the debugger I can dump only a few signals. Which signals are needed to understand why the Tilelink Hub deadlocks?

Dolu1990, 2025-03-20:
Hi,

haltRequest of l311 indicates that one of the previous transaction which colide with the current one is still in flight. Potentialy, it is one which happend before the wave capture ?

One thing to be very carefull about is the handeling of the source (axi ID) signal.

That is where i would look first. Because if that one is messed up, this will lock up the system in this way.

Maybe, setup you trigger on the hub io_down for a transaction with a non-zero source value, to check it comes back right.

petrmikheev, 2025-03-20:
Found previous write transaction to 0x80004240. It is followed by read at 0x80006240, and responses come in the opposite order, first read data, then write ack.

Looks correct?

[image]

petrmikheev, 2025-03-20:
> haltRequest of l311 indicates that one of the previous transaction which colide with the current one is still in flight. Potentialy, it is one which happend before the wave capture ?

Last transactions before the deadlock are all 64-byte reads with source=0, same as the ones visible in the dump.

Does the hub has something like a counter of in flight transactions? It would allow to find the place where something starts and not finishes.

petrmikheev, 2025-03-23:
I found the bug! Yes, it was incorrect `d_payload_source` in some cases. Capturing `hub_1/slots_*` helped to find the place where transaction didn't finish.

closed 2025-03-23 by petrmikheev