SpinalHDL/SpinalHDL #1688: Invalid stimulus data occurs during cross-clock domain simulation

buloruabutata, 2025-03-21:
In the second fork block, the pop signal should change with the for loop, but it is always 0 in the waveform.

[image]

```
SimConfig
    .withWave
    .withIVerilog
    .compile(AsyncFifo(8, 16)) // 实例化AsyncFifo模块
    .doSim { dut =>
		
		dut.popClockDomain.forkStimulus(10) 
		dut.pushClockDomain.forkStimulus(20) 
		dut.clockDomain.forkStimulus(10) 

		dut.io.push #= false 
		dut.io.pop #= false 
		dut.clockDomain.waitSampling(1)
		fork {
			dut.io.wresetn #= false
			dut.pushClockDomain.waitSampling(2) 
			dut.io.wresetn #= true
			dut.pushClockDomain.waitSampling(2) 
			for (i <- 0 until 100) {
				dut.io.data_in #= i // 设置要写入的数据
				dut.io.push #= true // 使能写操作
				dut.pushClockDomain.waitSampling()
				dut.io.push #= false // 禁用写操作
				dut.pushClockDomain.waitSampling()
			}
		}

		fork {
			dut.io.rresetn #= false
			dut.popClockDomain.waitSampling(2) // 等待2个读时钟周期
			dut.io.rresetn #= true
			dut.popClockDomain.waitSampling(2) // 等待2个读时钟周期，确保复位完成
			for (i <- 0 until 100) {
				dut.io.pop #= true
				dut.popClockDomain.waitSampling()
				dut.io.pop #= false
				dut.popClockDomain.waitSampling()
			}
		}

		dut.clockDomain.waitSampling(100)

		simSuccess()
	}
```

Dolu1990, 2025-03-23:
Hi,
This is very weird,
Can you provide the AsyncFifo component so we can reproduce in the exact same setup ?
Also, would need to know the iverilog exact version

I guess the bug isn't present if you keep the default simulator ? (verilator)

Thanks

buloruabutata, 2025-03-24:
Icarus Verilog version 12.0 (stable)

```
case class AsyncFifo(width: Int=4, depth: Int=8) extends Component {
	val io = new Bundle {
		val wclk = in Bool()
		val rclk = in Bool()
		val wresetn = in Bool()
		val rresetn = in Bool()
	    val data_in = in Bits(width bits)
		val data_out = out Bits(width bits)
		val push = in Bool()
		val pop = in Bool()
		val full = out Bool()
		val empty = out Bool()
	}

	noIoPrefix()

	val ptr_width = log2Up(depth) + 1

	val memory = Mem(Bits(width bits), depth)
	val clk_config = ClockDomainConfig(clockEdge=RISING, resetKind=SYNC, resetActiveLevel=LOW)
	val pushClockDomain = ClockDomain(io.wclk, io.wresetn, config=clk_config)
  	val popClockDomain = ClockDomain(io.rclk, io.rresetn)

	val push_var = new ClockingArea(pushClockDomain) {
		val wptr = Reg(UInt(ptr_width bits)) init(0) addTag(crossClockDomain)
		val rptr_gray_delay = Vec.fill(2)(Reg(UInt(ptr_width bits)) init(0) addTag(crossClockDomain))
	}

	val pop_var = new ClockingArea(popClockDomain) {
		val rptr = Reg(UInt(ptr_width bits)) init(0) addTag(crossClockDomain)
		val wptr_gray_delay = Vec.fill(2)(Reg(UInt(ptr_width bits)) init(0) addTag(crossClockDomain))
		val read_data = Reg(Bits(width bits)) init(0) addTag(crossClockDomain)
	}

	val rptr_gray = pop_var.rptr ^ (pop_var.rptr |>> 1)
	val wptr_gray = push_var.wptr ^ (push_var.wptr |>> 1)

	val push_logic = new ClockingArea(pushClockDomain) {
		when(io.push & ~io.full) {
			push_var.wptr := push_var.wptr + 1
			memory.write(push_var.wptr(ptr_width - 2 downto 0), io.data_in)
		}

		push_var.rptr_gray_delay.head := rptr_gray
		push_var.rptr_gray_delay.last := push_var.rptr_gray_delay.head

	}

	val pop_logic = new ClockingArea(popClockDomain) {
		when(io.pop & ~io.empty) {
			pop_var.rptr := pop_var.rptr + 1
			pop_var.read_data := memory.readAsync(pop_var.rptr(ptr_width - 2 downto 0))
		}
		pop_var.wptr_gray_delay.head := wptr_gray
		pop_var.wptr_gray_delay.last := pop_var.wptr_gray_delay.head
	}

	io.full := wptr_gray.asBits === Cat(~push_var.rptr_gray_delay.last(ptr_width - 1 downto ptr_width - 2), push_var.rptr_gray_delay.last(ptr_width - 3 downto 0))
	io.empty := rptr_gray.asBits === Cat(pop_var.wptr_gray_delay.last)
	io.data_out := pop_var.read_data

}
```

This problem can be solved by simulating in the following way

```
	SimConfig
	.withWave
	.withIVerilog
	.compile(AsyncFifo(8, 16))
	.doSim { dut =>

		dut.clockDomain.forkStimulus(10) 

		
		dut.io.push #= false
		dut.io.pop #= false
		dut.clockDomain.waitSampling(1)

		fork {
			dut.popClockDomain.forkStimulus(10)
		}

		fork {
			dut.pushClockDomain.forkStimulus(20)
		}

		fork {
			dut.io.wresetn #= false
			dut.clockDomain.waitSampling(2)
			dut.io.wresetn #= true
			dut.clockDomain.waitSampling(2) 
			for (i <- 0 until 100) {
				dut.io.data_in #= i 
				dut.io.push #= true
				dut.clockDomain.waitSampling(2)
				dut.io.push #= false 
				dut.clockDomain.waitSampling(2)
			}
		}

		fork {
			dut.io.rresetn #= false
			dut.clockDomain.waitSampling() 
			dut.io.rresetn #= true
			dut.clockDomain.waitSampling(20) 
			for (i <- 0 until 100) {
				dut.io.pop #= true // 使能读操作
				dut.clockDomain.waitSampling()
				dut.io.pop #= false // 禁用读操作
				dut.clockDomain.waitSampling()
			}
		}

		dut.clockDomain.waitSampling(100) // 等待50个读时钟周期

		simSuccess()
```

Dolu1990, 2025-03-25:
I can confirm the same bad behaviour on iverilog 11.0
So likely an issue with the SpinalHDL sim backend for iverilog.